prusa3d/PrusaSlicer #14611: 'Update check failed' keeps popping up every time there is no network connection to the prusa servers

Gregor-Gregor, 2025-06-02:
### Description of the bug

1. Every launch PrusaSlicer announces alarmingly 'Update check failed', meaning: the Prusa servers are not within reach.

[image]

That's not an error! There's no network connection on purpose (imagine that, some _isolated_ networks still exist). Internet access for selected apps (like PrusaSlicer) is only granted occasionally, under controlled conditions.
2. There's no option to silence this popup. I'm not speaking about disabling the update checks, they are very welcome. Just to get rid of this annoying popup.

### Project file & How to reproduce

To reproduce start PrusaSlicer without an internet connection.

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Win 11

### Printer model

any

Tupson444, 2026-09-19:
close only if not present in 3.0 (it is still in alpha so this doesn't show, but we don't know about final release yet)

Jan-Soustruznik, 2026-10-21:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion #16141.

closed 2026-10-21 by Jan-Soustruznik