prusa3d/PrusaSlicer #14629: Preview colors get gray when disabling feaure types

nomike, 2025-06-06:
### Description of the bug

When I'm in preview in the "Feature Type"  view, and deselect some of the feature types, the remaining ones turn grey at some point.

### Project file & How to reproduce

1. Start an empty project and add a sphere from the gallery.
2. Enable snug supports.
3. Open the preview render. - Notice that supports for example are colored green

[image]

4. Click on "Perimeter", "External perimeter", "Top solid infill" in the legend, to remove them from the preview

[image]

At that point all the remaining elements of the preview lose color and are just gray.

There is no one single option which when disabled triggers this behavior. It requires some combination or maybe a certain threshold.

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Ubuntu 25.04

### Printer model

n/a

u89djt, 2025-06-07:
(From a user taking an interest)
It looks like the display goes to black lines if the current layer according to the vertical slider doesn't contain anything to render.
[Shape-Sphere.zip](https://github.com/user-attachments/files/20638599/Shape-Sphere.zip)
[image]

Jan-Soustruznik, 2025-06-23:
Hello, @nomike, Thank you for your report. This issue has already been reported, so I’m closing it as a duplicate of #13285.
Thank you again!

closed 2025-06-23 by Jan-Soustruznik