prusa3d/PrusaSlicer #14630: Add First Layer Flow Rate Control in Print Settings Tab

stickyvinyl, 2025-06-07:
Hi PrusaSlicer team!

I’ve been using PrusaSlicer 2.9.2 for my PETG prints, and I’ve found that adjusting the first layer flow rate is essential for getting good adhesion and avoiding under- or over-extrusion issues. Currently, there’s no dedicated option in the Print or Filament Settings tabs to control First Layer Flow Rate directly—it’s only possible via live M221 commands or tweaking extrusion width.

It would be incredibly helpful if you could add a “First Layer Flow Rate” setting (similar to the existing first layer height and speed controls). This would make it much easier to fine-tune first layer adhesion across different materials without relying on manual overrides or post-processing scripts.

Thank you for considering this! It would be a fantastic quality-of-life upgrade for many users like me.

Cheers,
Scott

S3D-r, 2026-09-14:
[stickyvinyl], "tweaking extrusion width" doesn't affect the flow factor, so that it's not remedy for the problem.

Jan-Soustruznik, 2026-09-30:
Hello, @stickyvinyl, thank you for update, According to new policy I closing this feature request and converting it into a Discussion thread to hear what the community thinks about. Thank you, and have a nice day.