prusa3d/PrusaSlicer #14632: Possible Bug Command Line --repair & --merge switches no longer work

Soh5757, 2025-06-07:
### Description of the bug

Possible Bug Command Line --repair & --merge switches no longer work
Hi there! Forgive me if I am missing anything I'm supposed to have. I've never used the Prusa GitHub before. I checked for others with this problem but found no one.

I upgraded to 2.9.1 from 2.9.0 and found that a python script I created to render stl models using the Prusa command line engine stopped working. I found that I would get an actual error output from on my console saying --repair was not recognized. Commenting out --repair I would no longer get an error at console, but I would during debugging with the error below. I noticed that "returned non-zero exit status [3221225477](tel:3221225477)" always returned this address. From what I can tell, it's just some address that isn't within the read area allotted to Prusa. Outside of that, I have no idea. So, either I can remove the --merge switch, which then Mutiple models no longer merge, or if I uninstall and go back to 2.9.0, everything goes back to working. 

I did read the release notes and saw there was some changes to auto arranging prints for printing whole parts one at a time, so it made sense to me maybe something got misconfigured during this new feature. Just a total guess though. I wanted to report this incase this was a legit bug. 

Exception has occurred: CalledProcessError Command '['C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer-console.exe', '--gcode', '--merge', '--scale', '100%', '--load', 'D:\Users\Soh57\Documents\DND Mercant models\Automation\Etsyapp\Profiles\default\filament_Default.ini', '--load', 'D:\Users\Soh57\Documents\DND Mercant models\Automation\Etsyapp\Profiles\default\printer_Default.ini', '--load', 'D:\Users\Soh57\Documents\DND Mercant models\Automation\Etsyapp\Profiles\default\print_Default.ini', '--output', 'D:\Users\Soh57\Documents\DND Mercant models\Patreon\Black Scroll Gaming\Portals\309001_BSG_Calling_Portals_01_Enchanted_Glen\STLs\test_output.gcode', 'D:\Users\Soh57\Documents\DND Mercant models\Patreon\Black Scroll Gaming\Portals\309001_BSG_Calling_Portals_01_Enchanted_Glen\STLs\309001_EG_top_front.stl']' returned non-zero exit status [3221225477](tel:3221225477). File "D:\Users\Soh57\Documents\DND Mercant models\Automation\Etsyapp\Scripts\test2.py", line 21, in main result = subprocess.run(command, capture_output=True, text=True, check=True) File "D:\Users\Soh57\Documents\DND Mercant models\Automation\Etsyapp\Scripts\test2.py", line 34, in main() subprocess.CalledProcessError: Command '['C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer-console.exe', '--gcode', '--merge', '--scale', '100%', '--load', 'D:\Users\Soh57\Documents\DND Mercant models\Automation\Etsyapp\Profiles\default\filament_Default.ini', '--load', 'D:\Users\Soh57\Documents\DND Mercant models\Automation\Etsyapp\Profiles\default\printer_Default.ini', '--load', 'D:\Users\Soh57\Documents\DND Mercant models\Automation\Etsyapp\Profiles\default\print_Default.ini', '--output', 'D:\Users\Soh57\Documents\DND Mercant models\Patreon\Black Scroll Gaming\Portals\309001_BSG_Calling_Portals_01_Enchanted_Glen\STLs\test_output.gcode', 'D:\Users\Soh57\Documents\DND Mercant models\Patreon\Black Scroll Gaming\Portals\309001_BSG_Calling_Portals_01_Enchanted_Glen\STLs\309001_EG_top_front.stl']' returned non-zero exit status [3221225477](tel:3221225477).

### Project file & How to reproduce

 Use command line switch —repair of —switch 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Windows 11

### Printer model

Ender 3 v3 se

YuSanka, 2025-06-16:
Thanks for your report.
Could you please provide a more detailed description of the issue?
It would be very helpful if you could share step-by-step instructions to reproduce the problem, or provide your script.

Also, we’d like to point out that the `--repair` operation has been removed from the CLI, as models have been repaired by default for quite some time. The command was redundant and has therefore been removed from the command list.

However, the `--merge` command should still work as expected, its behavior has not been changed.

Jocarnail, 2025-06-22:
I'm having the same issue with `--merge`. Slicer just crash.
Here is the crash report:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [13210]
Path:                  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.9.2 (PrusaSlicer-2.9.2)
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [13207]
Responsible:           kitty [1409]
User ID:               501

Date/Time:             2025-06-22 16:33:14.2967 +0200
OS Version:            macOS 15.0.1 (24A348)
Report Version:        12
Anonymous UUID:        9DD115B1-384B-2C26-8652-35E2CC356661

Time Awake Since Boot: 5900 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [13210]

VM Region Info: 0 is not in any region.  Bytes before following region: 4362141696
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104010000-1085d8000    [ 69.8M] r-x/r-x SM=COW  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   PrusaSlicer                   	       0x104046124 Slic3r::CLI::process_transform(Slic3r::CLI::Data&, Slic3r::DynamicPrintConfig const&, std::__1::vector<Slic3r::Model, std::__1::allocator<Slic3r::Model>>&) + 740
1   PrusaSlicer                   	       0x104064c74 Slic3r::CLI::run(int, char**) + 520
2   dyld                          	       0x183878274 start + 2840

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000000   x1: 0x000060000259c3a4   x2: 0x0000000000000001   x3: 0x0000000000000000
    x4: 0x0000000000000064   x5: 0x000000000000006d   x6: 0x0000000127f05470   x7: 0x00000001083aaf1e
    x8: 0x000000016bdeede8   x9: 0x000060000259c3a0  x10: 0x746100656772656d  x11: 0x0000000000000170
   x12: 0x00000000000007fb  x13: 0x00000000000007fd  x14: 0x00000000dd36a018  x15: 0x00000000dd169832
   x16: 0x0000000183c2d950  x17: 0x0000000000000018  x18: 0x0000000000000000  x19: 0x7272615f746e6f64
   x20: 0x0000600000985a40  x21: 0x000000016bdeed50  x22: 0x0000000000000000  x23: 0x000060000328c238
   x24: 0x0000000183872000  x25: 0x0000000000000000  x26: 0x0000000000000000  x27: 0x000000010838a422
   x28: 0x000000016bdeed38   fp: 0x000000016bdeec30   lr: 0x0000000104046120
    sp: 0x000000016bdee8f0   pc: 0x0000000104046124 cpsr: 0x60001000
   far: 0x0000000000000000  esr: 0x92000006 (Data Abort) byte read Translation fault

Binary Images:
       0x104010000 -        0x1085d7fff com.prusa3d.slic3r/ (PrusaSlicer PrusaSlicer-2.9.2) <4f29cdc1-f221-357e-baaf-69b8fdbaca43> /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
       0x183872000 -        0x1838f4663 dyld (*) <40bedd20-da23-3b70-a9c2-a1c68a738762> /usr/lib/dyld
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???
       0x183c2c000 -        0x183c33fe3 libsystem_platform.dylib (*) <3fac373b-5c41-3c83-8f9f-c9e781e887da> /usr/lib/system/libsystem_platform.dylib

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
[... 170 lines cut ...]
  ]
}
}

Model: MacBookPro18,3, BootROM 11881.1.1, proc 10:8:2 processors, 32 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Jul 26 2024 20:03:40 version 20.10.1123.2.8.7.186 FWID 01-94146b8e
IO80211_driverkit-1302.71 "IO80211_driverkit-1302.71" Aug  5 2024 22:09:56
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB3.0 Hub
USB Device: USB 10/100/1000 LAN
USB Device: USB2.0 Hub
USB Device: Gaming Mouse G502
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

```

FelipeCybis, 2025-10-28:
Having same issue with `--merge` at PrusaSlicer-2.9.3 on Windows 11

This is the whole output of running at `--loglevel 5`:
```pwsh
& 'C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer-console.exe' 'C:\Program Files\Prusa3D\PrusaSlicer\resources\shapes\3DBenchy.stl' 'C:\Program Files\Prusa3D\PrusaSlicer\resources\shapes\box.stl' --loglevel 5 --merge --gcode --output test.gcode
```
```console
[2025-10-28 19:57:00.767431] [0x0000f278] [debug]   TriangleMesh::repair() started
[2025-10-28 19:57:00.818609] [0x0000f278] [debug]   TriangleMesh::repair() finished
[2025-10-28 19:57:00.917806] [0x0000f278] [debug]   TriangleMesh::repair() started
[2025-10-28 19:57:00.917806] [0x0000f278] [debug]   TriangleMesh::repair() finished
```
And then it just ends with no slicing.

lukasmatena, 2026-09-01:
Fixed in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena