prusa3d/PrusaSlicer #14634: Display used filament total in gcode viewe

TenaciousLemur, 2025-06-08:
**Is your feature request related to a problem? Please describe.**
Sometimes I need to know how much filament a print used and I would rather not remember/refill the parameters and re-slice.

**Describe the solution you'd like**
Solution: Add the total to the bottom of the list in the G-code viewer, the rest is already there and I could calculate it myself but this seems an elegant and simple feature for saving time and it already exists in the slicer proper.

**Describe how it would work**
Just a simple sum displayed as the last item in the list maybe with a little dividing line.

**Describe alternatives you've considered**
Use calculator or re-slice with same settings.

**Additional context**
None

neophyl, 2025-06-08:
Just open your gcode with a normal text editor,  towards the end of all gcode files produced by PS is that data already.

For example
; filament used [mm] = 1825.70
; filament used [cm3] = 4.39
; filament used [g] = 5.45
; filament cost = 0.16
; total filament used [g] = 5.45
; total filament cost = 0.16
; estimated printing time (normal mode) = 35m 20s
; estimated printing time (silent mode) = 35m 26s
; estimated first layer printing time (normal mode) = 1m 35s
; estimated first layer printing time (silent mode) = 1m 36s

Its located just after the end gcode block and before the list of all the settings used to produce the slice.

HSBallina, 2025-06-08:
If you switch to the `Tool` view you can see the total per extruder.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]