prusa3d/PrusaSlicer #14639: Start G-code ignores extrusion multiplier

WhiterRice, 2025-06-11:
### Description of the bug

[image]

[image]

[image]

This has been observed by a community member 9 months ago here, https://help.prusa3d.com/article/extrusion-multiplier-calibration_2257

### Project file & How to reproduce

Start a new project, set the filament extrusion multiple, slice, export gcode. View gcode.

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11 Pro

### Printer model

MK3S+

rtyr, 2025-06-11:
That is correct. Extrusion multiplier and firmware flow factor (M221) are two separate things.

closed 2025-06-11 by rtyr

WhiterRice, 2025-06-11:
My mistake, thanks for explaining.