prusa3d/PrusaSlicer #14646: Core ONE Chamber Minimal Temperature not honored

BootsSiR, 2025-06-14:
### Description of the bug

PrusaSlicer 2.9.2
Printing on a Core ONE

Slicing with the following chamber temperature settings:

[image]

Printer is above minimal temperature but print still does not start until nominal temperature reached:

[image]

### Project file & How to reproduce

[core_one_lid_magnets-Corner.zip](https://github.com/user-attachments/files/20738451/core_one_lid_magnets-Corner.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11

### Printer model

Core ONE 6.3.3+10442

BootsSiR, 2025-06-14:
@rtyr sorry, based on my use so far I thought that 'absorbing heat' was meant to indicate the chamber was absorbing the heat to reach nominal temperature.  Is there a documented explanation for what this phase is actually doing somewhere?

rtyr, 2025-06-14:
It basically ensures that the heat from the heatbed is absorbed to the heatbed and it's platform before the print rather than during the print. That is also the reason why this phase takes different amount of time based on heatbed temperature.

BootsSiR, 2025-06-14:
> It basically ensures that the heat from the heatbed is absorbed to the heatbed and it's platform before the print rather than during the print. That is also the reason why this phase takes different amount of time based on heatbed temperature.

Ahhhh, that makes total sense to me now.  Ensures the entire printer is 'temperature stable' before printing.  Thanks for the education!