prusa3d/PrusaSlicer #14663: 3DCONNEXION slow left right

bobbyh41, 2025-06-21:
### Description of the bug

On mac 15.5 and with the latest 3d connexion driver 10.8.8 i have to turn the mac sensitivity for the prusaslicer app all the way up on speed and that allows it to zoom and tilt properly, but the left and right is painfully slow. This has been a problem for a while, but I just updated everything hoping it would fix it and it did not.

### Project file & How to reproduce

It is a problem if a project is being worked on or not

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Mac 15.5

### Printer model

Any Printer

u89djt, 2025-06-21:
(fellow spacemouse user) Does enabling legacy support do anything different? Bottom line in general preferences.

bobbyh41, 2025-06-21:
> (fellow spacemouse user) Does enabling legacy support do anything different? Bottom line in general preferences. 

No, it does not help any

u89djt, 2025-07-02:
I just plugged in an older spacemouse and that gives painfully slow left/right under windows too, even with overall speed gain at max and that axis at max under advanced setting.
I've been using one of the recent wireless spacemouse devices, so 3dconnexion are interested in that working, and it does work fine in the slicer.
My reading is that 3dconnexion haven't helped developers mange their deprecated devices. I don't what type you have.
This is not a recommendation, just flagging a potential concept if you felt you could do it safely: could installing an old version of the 3dconnexion driver change things?

bobbyh41, 2025-07-03:
> A developer might chip in, but I would be surprised.
> For my own exploration, could I ask you what model/serial/date of spacemouse you have? If I happen to find a solution, I'll drop it here, but I'm not optimistic. I'm mostly looking for search inspiration.
> 

I can tell you from looking at my Amazon order from 2020 that the part number is 3DX-700059, I will have to try to find the serial number later, I appreciate your interest in this.

u89djt, 2025-07-03:
FYI I've installed the most recent driver listed as compatible with my ancient SpaceNavigator from the 3dconnextion support pages and the speeds are all good in prusaslicer and fusion. That doesn't help you with yours, which seems to be up to date. I'll report back when I've had a chance to experiement with the new one.
A reminder that my good experiences have been with the wireless spacemouse. I don't know if that will make a difference.

bobbyh41, 2025-07-03:
> FYI I've installed the most recent driver listed as compatible with my ancient SpaceNavigator from the 3dconnextion support pages and the speeds are all good in prusaslicer and fusion. That doesn't [...]

Ok, thanks

u89djt, 2025-07-05:
OK, the 3DX-700059 that just arrived works perfectly under Windows 11 with driver 10.9.6. I'll be able to test it on a mac laptop from a couple of years ago early next week, I think.

bobbyh41, 2025-07-06:
> OK, the 3DX-700059 that just arrived works perfectly under Windows 11 with driver 10.9.6. I'll be able to test it on a mac laptop from a couple of years ago early next week, I think.

Thanks. I keep digging in the setting and the only thing I can find to adjust related to speed is all or nothing, no choice for just the left/right. The only other program I use the mouse in is fusion and it works normal in that.

u89djt, 2025-07-06:
Ah now that's interesting. In some pieces of software, the settings dialog has a number of tabs on the right of the advanced pane, and it doesn't necessarily default to setting speed, so you have to click on speed to expose the sliders. Is that relevant for you?

bobbyh41, 2025-07-06:
> Ah now that's interesting. In some pieces of software, the settings dialog has a number of tabs on the right of the advanced pane, and it doesn't necessarily default to setting speed, so you have to click on speed to expose the sliders. Is that relevant for you?

I can select prusaslicer and it's just one big slider for speed for everything, I can't adjust individual axis's. There are more tabs for setting up what buttons and stuff do, but just one speed.

u89djt, 2025-07-06:
Can you take a screenshot of what you see when you hit the "Advanced" button?

bobbyh41, 2025-07-06:
> Can you take a screenshot of what you see when you hit the "Advanced" button?

I must be overlooking the advanced button.

[image]

[image]

u89djt, 2025-07-06:
What happens when you select one of the axes in the list in that second screenshot?
This is a totally different settings dialog to what we get under windows, so we're in frobnication territory!

closed 2026-09-29 by prusaslicer-bot[bot]