prusa3d/PrusaSlicer #14668: Prusa Slicer Under-extrusion after many small fills.

juckendesAuge, 2025-06-25:
### Description of the bug

Under-extrusion after many small fills.
Unfortunately, Prusa Slicer 2.9.2 scores many holes in a row and then prints the perimeters. The perimeters are under-extruded at the seam.

See video and photos attached.

### Project file & How to reproduce

[holder.3mf.zip](https://github.com/user-attachments/files/20910567/holder.3mf.zip)

https://github.com/user-attachments/assets/676bd178-7934-4ea1-9f23-5cbf6ff3f0f2
[image]
[image]
[image]
[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.2 (Flatpak)

### Operating system

Nobara release 42 (Forty Two)

### Printer model

Original Prusa MK3.5S,  Firmware 6.2.4+8909

Gik-0101, 2025-06-26:
+1

I've been trying to point out this error for a long time, but it's falling on deaf ears.

I've had this error for three or four years on my Voron 2.4 and now also on my Prusa XL.

After small extrusions (or gap fillings), there's always a lack of material at the perimeter start. This is only noticeable if it's on the outer perimeter. The "plaster" retraction extra length fixes this, but it causes some overextrusion elsewhere.

No, I haven't tested any other slicers like Cura, Orca, etc.

juckendesAuge, 2026-09-18:
Over time, I've reported about 13 bugs. Many weren't addressed or even looked into at all.
Now they're being closed because a new version is out, without checking whether the problem has been fixed.
This approach teaches me only one thing.
Submitting bugs is a waste of time. It takes a lot of time and isn't appreciated at all.
The only thing you’ve achieved is that I won’t be spending any more time reporting new bugs here.
Thanks a lot for your “simple solution”!