prusa3d/PrusaSlicer #14670: Ssl Connection error

AvariCe-git, 2025-06-25:
### Description of the bug

When PrusaSlicer starts, it shows this despite being logged in

[image]

When I try to do it manually, I get this error

[image]

What should I be doing to resolve this?

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11

### Printer model

Prusa mk4s

AvariCe-git, 2025-06-27:
> Hello, [@AvariCe-git](https://github.com/AvariCe-git) ,are you trying to connect from the company network?

No, I am from home. I am however behind a pi hole, though I do not see any relevant connection block. Is there a domain that I should unblock?

Jan-Soustruznik, 2025-06-30:
Hello, @AvariCe-git, That's strange, these ERR messages are often related to issues with a company network. Try this solution: 

What is happend(probably -- accorfing to ERR message:)
When you login in the embedded browser window, Prusa Slicer tries to communicate with https://account.prusa3d.com/ which fails as it is not able to verify if the HTTPS certificate of [acconut.prusa3d.com](http://acconut.prusa3d.com/) is not revoked (it tries to do the request which is blocked likely). this is likely related to some corporate IT stuff like VPN or security software (antivirus/antimalware)...
the correct solution is to ask the IT department to whitelist PrusaSlicer and/or ***.prusa3d.com** domains to allow Prusa Slicer to access to.(including the HTTP protocol for google servers ***.pki.goog** that are used to check the revoked certificates)

Note: if wondering why the embedded browser works with HTTPS at account.prusa3d.com but the PrusaSlicer is not, it is  because the embedded browser is identified as Edge browser while the PrusaSlicer is different process... so the whitelisting treats the browser and PrusaSlicer as two distinct applications and hence PS can be blacklisted while its internal browser is not.

Let me know if help. thank you. have a nice day

AvariCe-git, 2025-06-30:
That was it, I had Edge and msedgewebview2 blocked locally. As soon as I unblocked them and logged on, it was able to connect and download the updates.