prusa3d/PrusaSlicer #14687: changement de remplissage

tobiaslive, 2025-07-03:
### Description of the bug

lorsque je veux changer le remplissage entre 2 hauteurs 
je fait "ajouter un modificateur de plage" 
je séclectionne les hauteur min et max 
et je choisi le remplissage et le type de remplissage
je fait redécoupage 

mais je ne vois aucun changement au niveau du remplissage lorsque je me déplace dans les hauteur de mon objet 

### Project file & How to reproduce

[embout v2.zip](https://github.com/user-attachments/files/21045743/embout.v2.zip)

[embout v2.zip](https://github.com/user-attachments/files/21045753/embout.v2.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer Version:   2.9.2 Build:     PrusaSlicer-2.9.2

### Operating system

Operating System:    Windows System Architecture: 64 bit Windows Version:     Windows 10 (version 19045), édition 64-bit Total RAM size [MB]: 8,490MB OpenGL installation GL version:   3.3.0 (3.3 (Core Profile) Mesa 12.0.0-rc2) Profile:      Core Vendor:       VMware, Inc. Renderer:     Gallium 0.4 on llvmpipe (LLVM 3.6, 256 bits) GLSL version: 3.30.0

### Printer model

MKS3+

u89djt, 2025-07-03:
Il faut spécifier la plage de hauteur en mm, et non en couches, comme ceci:
[a mm embout v2.zip](https://github.com/user-attachments/files/21045933/a.mm.embout.v2.zip)
[image]

u89djt, 2025-07-03:
Il s'agit simplement d'un autre utilisateur qui suit le référentiel - cela leur fait gagner du temps :)