prusa3d/PrusaSlicer #14690: Unable to build deps on MacOS

pRizz, 2025-07-07:
### Description of the bug

I followed the steps in docs/How to build - Mac OS.md but I am getting a handful of build errors when running `make`, specifically related to 

```
PrusaSlicer/deps/build/dep_Blosc-prefix/src/dep_Blosc/internal-complibs/zlib-1.2.8/gzread.c:30:15: error: call to undeclared function 'read'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
   30 |         ret = read(state->fd, buf + *have, len - *have);
      |               ^
PrusaSlicer/deps/build/dep_Blosc-prefix/src/dep_Blosc/internal-complibs/zlib-1.2.8/gzlib.c:256:24: error: call to undeclared function 'lseek'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  256 |         state->start = LSEEK(state->fd, 0, SEEK_CUR);
PrusaSlicer/deps/build/dep_Blosc-prefix/src/dep_Blosc/internal-complibs/zlib-1.2.8/gzlib.c:355:9: error: call to undeclared function 'lseek'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  355 |     if (LSEEK(state->fd, state->start, SEEK_SET) == -1)
      |         ^
```

Do I need to configure a specific version of C or something?

### Project file & How to reproduce

Follow the steps in docs/How to build - Mac OS.md, until the make command

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

master

### Operating system

macOS 15.5

### Printer model

N/A

pRizz, 2025-07-07:
For posterity, I was able to get past the error: when installing `zlib` with `brew`, as specified in the doc, I realized it output this message:
```
For compilers to find zlib you may need to set:
  export LDFLAGS="-L/opt/homebrew/opt/zlib/lib"
  export CPPFLAGS="-I/opt/homebrew/opt/zlib/include"
```

Then, when I deleted the `build` folder, and reran `cmake` and `make`, there were no more `zlib` build errors.

I am getting a new error now:

```
-- Using src='https://gitlab.com/libeigen/eigen/-/archive/3.3.7/eigen-3.3.7.zip'
-- verifying file...
       file='PrusaSlicer/deps/build/downloads/Eigen/eigen-3.3.7.zip'
-- SHA256 hash of
    PrusaSlicer/deps/build/downloads/Eigen/eigen-3.3.7.zip
  does not match expected value
    expected: '4815118c085ff1d5a21f62218a3b2ac62555e9b8d7bacd5093892398e7a92c4b'
      actual: 'e09b89aae054e9778ee3f606192ee76d645eec82c402c01c648b1fe46b6b9857'
-- Hash mismatch, removing...
CMake Error at dep_Eigen-stamp/download-dep_Eigen.cmake:170 (message):
  Each download failed!
```

amatulic, 2025-09-02:
Did you ever get this working? I haven't been able to build PrusaSlicer on Windows since 2.9.1 (issues #14384 and #14573).

closed 2026-09-05 by pRizz