prusa3d/PrusaSlicer #14694: Single Extruder Multi Material M600 GCode ignored?

cschalle, 2025-07-07:
### Description of the bug

So I got an item with numbers on them and I want the numbers to be using a different color for the last layer. I assume Single Extruder Multi Material should help me do this.

So I import my STL file and then go into 'Printers' in PrusaSlicer. There I chose 2 extruders and click the Single Extruder Multi Material option. Then I go into each extruder and choose 2 different colors.

Then back in the plater I choose the multimaterial painting tool and then use tool type smartfil to paint the numbers using extruder color 2. 

Then I use prusaconnect to send the print to my CoreOne. So what I would expect to happen is that it stops before the final layer and lets me change filament. Instead it just finishes the print using the single filament.

On a Prusa forum thread it was suggested I add the M600 code to the Toolchange Gcode which seems to work, although now I am getting 'bonus' filament changes at various points. 

In the Color Change G-code section though I also see the M600 code which I understand is what triggers the filament change, but for some reason it never seem to trigger from there. 

### Project file & How to reproduce

[5x-lid-4-monsters.zip](https://github.com/user-attachments/files/21104342/5x-lid-4-monsters.zip)

Printing this file should generate a filement change for the numbers on the print, but currently it seems not to happen on my Core One.

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Fedora Workstation Linux

### Printer model

Prusa Core One

cschalle, 2025-07-07:
But I shouldn't have to manually add the M600 code to the toolchanger Gcode at all right? the one added automatically in the Color Change gcode section should suffice right?

neophyl, 2025-07-07:
No.  The gcode in the colour change field is used when you manually insert a colour change command using the layer slider after you have sliced.

Tool change is a completely separate process.  While the desired end result may be a different colour in this case, it could just as easily be a change to a different filament type etc.

closed 2026-09-29 by prusaslicer-bot[bot]