prusa3d/PrusaSlicer #14699: bug in flatpak

WOWElectronics, 2025-07-09:
### Description of the bug

2.8.1 appimage works, 2.9.2 flatpak doesn't. Followed install instructs for flatpak, then prusaslicer, now prusaslicer test's printer connection, says working properly, but click on physical-printer says klipper not running for 5 seconds and bails out. Wash, rinse, repeated many times.  Restore link to PS281, all seems to work.  firefox to localhost on printers own monitor is normal fluid display.  Why flatpak, its 50,000 files of pure bloat on a debian bookworm system.

### Project file & How to reproduce

AppImage 2.8.1 works is printing now. flatpak doesn't, goto cog in either, checks as "working properly" , load flatpak, click on cog, say its working properly, click on physical_printer, gets klipper not running msg, bails out totally 5 seconds later. Totally is quite right, have to hit enter to get shell prompt back.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2 flatpak on debian x8664 bookworm all up to date

### Operating system

debian 12 64 bit bookworm fully up to date

### Printer model

ender 5 plus, completely rebuilt with closed loop stepper/servo's lots of extra heat

rklxmok, 2025-07-16:
I am having this same issue on prusa-slicer 2.9.2-10 from the arch repo, and the flatpak version 2.9.2 - Debug Info Below

`Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
Downloading 139.95 K source file /usr/src/debug/webkit2gtk-4.1/webkitgtk-2.48.3/Source/WebKit/UIProcess/API/gtk/WebKitWebViewBase.cpp
webkitWebViewBaseEnterAcceleratedCompositingMode (webkitWebViewBase=<optimized out>, layerTreeContext=...)
at /usr/src/debug/webkit2gtk-4.1/webkitgtk-2.48.3/Source/WebKit/UIProcess/API/gtk/WebKitWebViewBase.cpp:2879
2879        webkitWebViewBase->priv->acceleratedBackingStore->update(layerTreeContext);`

Samueru-sama, 2025-07-20:
@WOWElectronics 

* https://github.com/prusa3d/PrusaSlicer/issues/13653 

* https://github.com/gneiss15/PrusaSlicer.AppImage

WOWElectronics, 2025-07-20:
Thank you, I have pulled both 2.9.1 and 2.9.2 but its the not so wee hours of the nite/ morning here, so its not yet tried. I have not been able to get the flatpak's to connect to the printer so I've been stuck on 2.8.1 for a while. Hopefully these AppImages will work ok. 

One battle that would allow better bridging, would be a place as an a "," separated list of layers that correspond to the layer number where a new bridge is started, a marker, where the nozzle temp is lowered just for that beginning layer, resulting in a higher tension to hold up the middle of a new bridge, then restore the nozzle temp to the usual higher temp for the next layer.  Seems to me that could improve bridges in printers normally moving several times their OOTB speeds.

closed 2026-09-29 by prusaslicer-bot[bot]