prusa3d/PrusaSlicer #14714: Set Right Mouse Button as a dedicated camera rotate

FatalStrings, 2025-07-17:
it is impossible to rotate the camera when there is no 'blank' space to left click into. it is especially annoying when you are zoomed in as the whole screen is occupied by the model.

At the moment the Middle Mouse button and Right click does the same thing, that is to Pan the camera. Since most mouse already has a middle mouse button anyway why not remove this redundancy?

in addition, the current controls put too much functionality to the left click that it makes the user experience infuriating. every time I rotate the camera there is a chance that I might move the model from the build plate, this gets annoying fast.

u89djt, 2025-07-17:
(From a fellow user) Hold the crl key down before and during your click and drag and that will rotate the disply even when there's no free space.

Tupson444, 2025-07-17:
>Hold the crl key

Thanks for this, I didn't know that.

But I would still like what FatalStrings suggested: to have a dedicated mouse key for rotating camera (perhaps use middle button for rotating and keep the right click for pan ?).

FatalStrings, 2025-07-18:
> (From a fellow user) Hold the crl key down before and during a left click and drag and that will rotate the camera even when there's no free space.

well damn... I didn't know that.