prusa3d/PrusaSlicer #14716: Prusa slicer stopped working on Ubuntu

busybeaver42, 2025-07-18:
### Description of the bug

Prusa Slicer don't show the 3D view inside the plater sheet. Start to slice kill the app. 
That happen for 2 days, after Nvidia driver update (gl issue ?). 
Another application like meshlab and blender works fine. 

My System info:
Ubuntu 24.04.2 LTS
SchenkerTechnologiesGmbH XMG NEO _E24_
Intel® Core™ i9-14900HX × 32
64.0 GiB RAM
NVIDIA GeForce RTX™ 4090 Laptop GPU

+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.163.01             Driver Version: 550.163.01     CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4090 ...    Off |   00000000:01:00.0  On |                  N/A |
| N/A   40C    P8              2W /  150W |     548MiB /  16376MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      3549      G   /usr/lib/xorg/Xorg                            184MiB |
|    0   N/A  N/A      3912      G   /usr/bin/gnome-shell                           93MiB |
|    0   N/A  N/A      5938      G   ...irefox/6495/usr/lib/firefox/firefox        221MiB |
|    0   N/A  N/A     11123      G   /usr/bin/nautilus                              17MiB |
|    0   N/A  N/A     13890      G   /usr/bin/gnome-control-center                  15MiB |
+-----------------------------------------------------------------------------------------+

Additional info:
I'm not the only one with this issue:
https://www.reddit.com/r/3Dprinting/comments/1c2gckl/comment/kza5d7e/

### Project file & How to reproduce

none

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.9.2

### Operating system

Ubuntu 24.04.2 LTS

### Printer model

Prusa Core One

Jan-Soustruznik, 2025-07-18:
Hello, @busybeaver42 , thank you for your info. YYes, we've seen some issues related to recent NVIDIA driver updates.. Try make a update.

> I had the same on my Debian testing bookwoork and NVIDIA driver 575.57.08. `flatpak update` Do the trick.
https://github.com/prusa3d/PrusaSlicer/issues/14616#issuecomment-2940499869

StephenFluin, 2025-07-19:
Had the same issue (slicer preview missing). `flatpak update` worked perfectly to fix it.

```$ flatpak update
Looking for updates…

Info: runtime org.gnome.Platform branch 46 is end-of-life, with reason:
   The GNOME 46 runtime is no longer supported as of April 17, 2025. Please ask your application developer to migrate to a supported platform.

        ID                                                            Branch                   Op              Remote              Download
 1. [✓] org.flathub.flatpak_external_data_checker.Locale              stable                   u               flathub               1.6 kB / 1.9 MB
 2. [✓] org.freedesktop.Platform.GL.default                           23.08                    u               flathub               7.5 MB / 169.7 MB
 3. [✓] org.freedesktop.Platform.GL.default                           23.08-extra              u               flathub               6.6 MB / 169.7 MB
 4. [✓] org.freedesktop.Platform.GL.nvidia-570-133-07                 1.4                      u               flathub               3.3 kB / 299.0 MB
 5. [✓] org.freedesktop.Platform.GL.nvidia-570-169                    1.4                      i               flathub             202.0 MB / 299.0 MB
 6. [✓] org.gtk.Gtk3theme.Breeze                                      3.22                     u               flathub               6.1 kB / 192.6 kB
 7. [✓] org.flathub.flatpak-external-data-checker                     stable                   u               flathub             372.5 kB / 18.6 MB

Changes complete.

        ID                                                           Branch             Op             Remote              Download
 1. [✓] org.flathub.flatpak_external_data_checker.Locale             stable             u              flathub               1.6 kB / 1.9 MB
 2. [✓] org.flathub.flatpak-external-data-checker                    stable             u              flathub             372.5 kB / 18.6 MB
 3. [✓] org.freedesktop.Platform.GL.nvidia-570-133-07                1.4                u              flathub               3.3 kB / 299.0 MB
 4. [✓] org.freedesktop.Platform.GL.nvidia-570-169                   1.4                i              flathub             202.0 MB / 299.0 MB
 5. [✓] org.gnome.Platform.Locale                                    48                 u              flathub              18.4 kB / 389.1 MB
 6. [✓] org.gtk.Gtk3theme.Breeze                                     3.22               u              flathub               6.1 kB / 192.6 kB
 7. [✓] org.gnome.Platform                                           48                 u              flathub               2.3 MB / 397.0 MB

Changes complete.

$ flatpak run com.prusa3d.PrusaSlicer
```

closed 2026-09-29 by prusaslicer-bot[bot]