prusa3d/PrusaSlicer #14725: Download from printer

HanyHad, 2025-07-23:
### Description of the bug

If I want to download a file from the printer, only a .tmp file is created on the disk, which is deleted after the slicer exits, but the file is not downloaded.

### Project file & How to reproduce

[70d54ca0-92aa-4634-add1-947c54b26cb4.zip](https://github.com/user-attachments/files/21396835/70d54ca0-92aa-4634-add1-947c54b26cb4.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11 Home

### Printer model

Prusa Core One

Jan-Soustruznik, 2025-07-24:
Hello, @HanyHad, Could you please clarify what you mean by "download from printer"? Are you referring to downloading through thePrusa Connect tab or PrusaLink tab in PrusaSlicer? It would also be helpful if you could provide some steps to reproduce the issue. Thank you, have a nice day

HanyHad, 2025-07-24:
@Jan-Soustruznik Dobrý den, pro mne lepší komunikace v CZ pokud to nevadí. Postup je následující, otevřu Průša  slicer, záložka connect, zvolím tiskárnu, vidím tiskové soubory uložené v tiskárně, výběru si soubor, kliknu v pravo na tři tečky, dám stáhnout. V pravém dolním rohu oznámení, že stahování začalo, na disku se vytvoří dočasný .tmp soubor a tím to skončí. Nikde jsem nenašel možnost sledovat průběh stahování a toto se ani nedokončí. I po několika hodinách po ukončení sliceru se odstraní .tmp soubor, ale ten tiskový co jsem chtěl stáhnout nikoliv. Zkoušel jsem spustit slicer normálně I s prací administrátora. Stejný průběh.