prusa3d/PrusaSlicer #14738: Prusa Slicer Configuration Wizard unable to open zip files

dan-p3rry, 2025-07-29:
### Description of the bug

I'm trying to add some filament profiles using the "Load..." button on third page of Configuration wizard.  But Wizard seems broken, refuses to open the zip file.  I tested the zip file by extracting the files, it's a valid zip file.  

Windows 11, PS 2.9.2

[Prusa XL config - PCTG.zip](https://github.com/user-attachments/files/21487379/Prusa.XL.config.-.PCTG.zip)

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.2

### Operating system

Windows 11

### Printer model

XL and Mk4s

dan-p3rry, 2025-07-30:
Where is this documented?  Why is the error message so useless?  You may as well remove the option from the wizard if it's implemented so poorly.  

Seriously ... the point of a "wizard" is it should be a user-friendly super-easy way to make changes.

Jan-Soustruznik, 2025-07-31:
There is no documentation on "How to create your own ZIP file" because this feature is intended as an alternative for offline PC users who, for any reason, do not have access to online sources. These offline ZIP files can be used to load printers, filaments, and print settings into PrusaSlicer.

[image]

I have closed the ticket as resolved.

closed 2025-07-31 by Jan-Soustruznik