prusa3d/PrusaSlicer #14748: Prusa connect login error

SorenThuesen, 2025-08-02:
### Description of the bug

Opening Prusa slicer, Prusa connect cannot login

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.2

### Operating system

WIN11

### Printer model

PRUSA MK4S

gwenaelnavez, 2025-08-02:
Found just a workaround on Mac OSX : just download again pruslaslicer 2.9.2 (actual version), login again and it's working.  So can you ry this on Windows?

kubispe1, 2025-08-02:
Hi, this is a duplicate. Please see issue #14746, and especially my comment here: [#14746 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/14746#issuecomment-3146551947). We will investigate it as soon as we are able to reproduce it

closed 2025-08-02 by kubispe1