prusa3d/PrusaSlicer #14749: Невозможно перевести эти строки

DrFOXUA, 2025-08-02:
### Description of the bug

#

### Project file & How to reproduce

Both compatible an incompatible presets are shown. Click to hide presets not compatible with the current printer.

Only compatible presets are shown. Click to show both the presets compatible and not compatible with the current printer.

Yes

No

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Win 64

### Printer model

all

DrFOXUA, 2025-08-02:
[image]

u89djt, 2025-08-03:
Насколько я вижу на скриншотах, толщина вашего объекта составляет 0,01 мм. В идеале, вам следует предоставить файл вашего проекта. Для этого его необходимо упаковать в zip-архив. Любой печатаемый слой заполняется пластиком только тогда, когда объект занимает более 50% своей высоты, поэтому вам нужно использовать высоту слоя менее 0,02 мм. Вряд ли ваш принтер сможет это сделать. И имейте в виду, что толщина слоя всё равно будет 0,02 мм, даже если это возможно. Вам придётся установить толщину 0,01 мм, чтобы получить пластик толщиной 0,01 мм. Пока это мечта!

As far as I can see from the screenshots, your object is 0.01 mm thick. Ideally, you should provide your project file. To do this, it needs to be packed in a zip archive. Any printed layer is filled with plastic only when the object takes up more than 50% of its height, so you need to use a layer height of less than 0.02 mm. It is unlikely that your printer will be able to do this. And keep in mind that the layer thickness will still be 0.02 mm, even if it is possible. You will have to set the thickness to 0.01 mm to get plastic that is 0.01 mm thick. For now, this is a dream!

DrFOXUA, 2025-08-03:
Это специально вызванная ошибка мною.
Тема о локализации. Я нашел строки которые невозможно перевести, так как они в сыром проекте не обозначены для Локализации
It is impossible to translate these lines. Not formar string gettexr - L or _L or _u8L

u89djt, 2025-08-03:
This is the user's issue "This is a deliberately caused error by me.
Topic about localization. I found strings that cannot be translated, since
they are not marked for Localization in the raw project"

DrFOXUA, 2025-08-05:
Я обращаюсь к создателям PrusaSlicer с просьбой о внесении этих строк для возможности перевода

DrFOXUA, 2025-08-05:
[image]
Еще одно строка для перевода

DrFOXUA, 2025-08-05:
Я пользуюсь и Тестирую Вашу PrusaSlicer. Далеко не все пользователи понимают Английский Язык, Для очень многих востребована Локализация на Свой Родной Язык...

DrFOXUA, 2025-08-19:
Повлиять только на разработчиков. Я еще накопал строк без перевода, но они обозначены в ресурсных файлах как для перевода, а в конечном итоге их нет. А есть и наоборот. Фактически все строки вытаскиваются по _L, L, _u8L  не считая _PLURAL.

closed 2026-09-29 by prusaslicer-bot[bot]