prusa3d/PrusaSlicer #14753: preview not avilable

Aaronfetamina, 2025-08-04:
### Description of the bug

when i slice i cant see the preview of the object (g code)

[image]

[image]

### Project file & How to reproduce

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

windows 11

### Printer model

prusa mini

Jan-Soustruznik, 2025-08-04:
Hello, @Aaronfetamina , could you share your project(3MF) file, please?

Aaronfetamina, 2025-08-04:
Hi, I updated the drivers of my GPU and reinstalled the slicer and it worked

Thanks!