prusa3d/PrusaSlicer #14758: PrusaSlicer keeps getting logged out of PrusaConnect

DavyGD, 2025-08-06:
### Description of the bug

Almost every time I use PrusaSlicer I get logged out of Connect and have to log in again. With 2FA enabled this takes a lot of effort to keep logging in a few times a day. This happens with me logging in through as well my Prusa account as logging through my Google account.

### Project file & How to reproduce

I cannot reproduce the problem on a project basis, I can not replicate it.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11 Pro

### Printer model

Prusa MK4s with MMU3

Crocmagnon, 2025-08-14:
I don't reproduce the issue on the Flatpak build.

jekkos, 2025-12-16:
Same problem for me, so annoying it makes me want to switch to another platform

closed 2026-09-29 by prusaslicer-bot[bot]