prusa3d/PrusaSlicer #14760: Fuzzy skin modifier

KAME-JW, 2025-08-07:
### Description of the bug

The fuzzy skin modifier to block the fuzzy skin in the middle also blocks fuzzy skin on surfaces outside the modifier volume.

### Project file & How to reproduce

[drift_planter_support case.zip](https://github.com/user-attachments/files/21668345/drift_planter_support.case.zip)

[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Win11

### Printer model

MK4S

neophyl, 2025-08-07:
Not entirely correct.  The modifier you are using for the infill in combination with the fuzzy skin blocker is the one blocking the fuzzy skin ion the main object walls for some reason.  If you move that modifier out the way or delete it then the fuzzy skin is generated where you want it.

[image]

If you move it so it just covers up to that interior top surface I believe you can get what you want with the current software.  Still a bug though :)

[drift_planter_support case_neo.zip](https://github.com/user-attachments/files/21671205/drift_planter_support.case_neo.zip)

KAME-JW, 2025-08-11:
Oh wow, thanks.

Best regards,
JW

image.png (view on web)<https://github.com/user-attachments/assets/3f2a62bd-3719-461f-b3cc-f789d2021ba0>

closed 2026-09-29 by prusaslicer-bot[bot]