prusa3d/PrusaSlicer #14764: Embossed seams

rahmanshaber, 2025-08-09:
**Is your feature request related to a problem? Please describe.**
Because of the seams the diameter of a hole becomes inaccurate.

**Describe the solution you'd like**
Embossed seams

**Describe how it would work**
Want an option in the slicer when the seams are created it will create a emboss. the emboss will a half cylinder shape with the half the diameter of current nozzle size.

for example, here the emboss diameter is 0.2mm for a 0.4mm nozzle

[image]

[image]

**Describe alternatives you've considered**
none

**Additional context**
none

Jan-Soustruznik, 2025-08-12:
Hello @rahmanshaber, have you tried the scarf seams?

rahmanshaber, 2025-08-12:
Hi, I did. It works well on well dried filament. But with more drupy filament like PETG it creates waste filament attached to the nozzle and later drops in the print and mess up the prints.

So I am stuck with old seams. Embossed seams would be the best solution.

u89djt, 2025-08-15:
Just dropping in some related motivation for managing deposition with concavities: Angus discussing the general issue of sharp internal corners, with the side benefit of a non-contacting concavity being that a seam can be hidden there: https://youtu.be/I52dAzPiIP8?si=LcDCsWfhloU-9zsp&t=522

rahmanshaber, 2025-08-16:
Hi, i did with the model pic I shared on top. but doing this manually for a big project is a lot.

u89djt, 2025-08-16:
Yeah, it's a bunch of work, but you can make punch bodies for all the sizes you need and use those to make holes with ready-made seam concealment.

Nickxsch, 2025-11-07:
I do like this idea, and with non-straight objects, doing this manually is imposs- very hard. Similar idea I had (and others too), is to make the perimeters one continuous line. So with three perimeters, the visible seam would be where it moves into (or out from) the first internal perimeter. This would also reduce retractions and drops elsewhere in your print.

rahmanshaber, 2026-09-02:
It's still relevant

u89djt, 2026-09-02:
Just realising that we should be calling them _de_bossed seams.
Or maybe "niche" seams. 
When the dust has settled around getting 3 to stable release, maybe they'll have time for discussion of whether this belongs as a slicer feature.

Jan-Soustruznik, 2026-09-30:
Hello, @rahmanshaber , thank you for update, According to new [policy](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I closing this feature request and converting it into a Discussion thread to hear what the community thinks about. Thank you, and have a nice day.