prusa3d/PrusaSlicer #14765: Please see the video I made which shows the issue as I describe it. This is a BIG issue with the XL purge tower IMO. Thanks https://drive.google.com/file/d/1ctAGkYzZM_NCTFQ7VIM7a8qAGTyaLr93/view?usp=sharing

Ozzy-Artist, 2025-08-10:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

u89djt, 2025-08-10:
(community triage)
I watched the user's video.
https://drive.google.com/file/d/1ctAGkYzZM_NCTFQ7VIM7a8qAGTyaLr93/view?usp=sharing
They state that tall wipe towers are routinely toppled by the tool crashing into buildup at sharp corners in the purge/equalization extrusion process.
Suggested title: "Tall wipe towers on XL routinely toppled by tool crashing into plastic buildup at sharp turns within the tower"
We don't have a project file.
To the user: post a zipped project file. There are plenty of folk eager to help, including finding workarounds or mitigations you might not have found yet.
I experience plastic buildup, too, but it hasn't toppled anything for me yet. 
The most common thing causing plastic buildup is at the start of an extrusion due to oozing caused byt moisture in the filament.
There was an issue posted recently where the user suggested changing the start point of the loops in the tower to avoid buildup - I had a quick look and failed to find it - someone will dig it out. It may be that this issue should be tied to that one.
Lots of people read these things and engage when there's enough information to make it an efficient use of their time.

Ozzy-Artist, 2025-08-10:
I cannot share the project because it contains a 3D model that is proprietary and protected under the license agreement I purchased. The PLA I use is dry. The oozing is caused by the filament sitting melted in the  extruder while it is parked - that's the whole purpose of purging the extruder a bit before printing. Also, I live in a non-humid region and I store my filament in sealed plastic bags with large desiccant pouches in them. This is a design flaw in the coding for the purge tower. What would work better is if the purges printed in more rounded loops with randomized start / stop points. 

The issue, as it is, is caused by the confined, narrow widths of the purge loops and the repeated start / stop points of the purges for each extruder.

Ozzy-Artist, 2025-08-10:
I know that but then you can't see the specifics of the purge tower. And posting the project is irrelevant. This issue is not project specific, or filament brand specific. Every tall, multicolour print I have attempted has failed due to the tower being dislodged from the plate due to the nozzle catching on blobs on the tower. The blobs are created by the repetitive start / stop points of the purge lines on the tower. 

If I could code, I would change the way the purge tower is printed...but I cannot code.

Ozzy-Artist, 2025-08-11:
Yes, this https://github.com/prusa3d/PrusaSlicer/issues/14529 , is exactly what I'm talking  about and he give the same solution I do. Attached is an original, unpublished* model of mine and a photo of the issue as I print this. Please do not save or print this model. I will publish it soon.
Model file:  https://drive.google.com/file/d/1OkAAWS_ViYa0cJg1AqB8R_n-vPU4aoyO/view?usp=sharing
[image]

u89djt, 2025-08-11:
If that's a project file you've posted, the only way to look at it would be to save it to disk first.

closed 2026-09-29 by prusaslicer-bot[bot]