prusa3d/PrusaSlicer #14767: Between Object Gcode happens after hotend move

TheAndr0id, 2025-08-10:
### Description of the bug

The overall problem I'm trying to solve is top layer surface scars when printing multiple objects.  The nozzle drags across the surface (sometimes of multiple objects) leaving a heat scar.

I thought if I was to use the Between Objects GCode (between_objects_gcode) to force the head to rise 5-10mm between objects to remove this issue. 

The problem is that gcode is only inserted AFTER the head moves to the new object.  That move should be part of the new object, not the current one.

Here's the current order:
1) Print object A
2) Move to object B start location
3) Insert between_objects_gcode
4) Move to Z=0 and start printing B

2 and 3 above should be reversed.  The move is part of the next object not the current one.  (For example, the last object has no move...)

### Project file & How to reproduce

1) Create two objects in the slicer (say Shape-Box)
2) Activate sequential printing, adjust spacing as needed for hotend spacing
3) Add a comment/code to between_objects_gcode
4) Slice and step through the gcode looking at travel and when the gcode in 3 is executed

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.2

### Operating system

Windows 7

### Printer model

Ender 3 v2 stock

TheAndr0id, 2025-08-17:
That's silly, but fine.  Here's a zip file with a project file and the STL exported from PrusaSlicer.  I assume the screenshots I provided above will suffice?  The generated g-code is attached in my previous message.

[Shape-Box.zip](https://github.com/user-attachments/files/21824695/Shape-Box.zip)

u89djt, 2025-08-19:
FWIW from me as a random user, I'm totally on board with the between-objects-gcode question.

I'm interested in coping with the situation when it eventually hits me or other folk while decisions are made about that code.

Re surface blobs, that suggests retraction/wipe settings could benefit from some attention? I assume you're effectively imposing a hard wipe on the next perimeter encountered by not rising during travel? You're presumably getting scoring inside your objects that you don't see at the end. It feels like there's a better answer.

Couple of thoughts so far:

Slightly laborious workaround: define a second extruder with travel lift and set the first and last layers to use that with height modifiers.

Simpler solution that might solve everything at once: this seems likely to be causing your blobs while printing:
[image]
Uncheck that, enable lift and you might be fine.

You're not using the Ender 3 V2 distributed with the slicer - you can add it as a preset from the Other FFF->Creality set. Try those defaults because the community behind it might have solved some of your problems for you.

u89djt, 2025-08-25:
A dev or mod might take the time to explain. I'm happy to go. away. :)
[image]

closed 2026-09-29 by prusaslicer-bot[bot]