prusa3d/PrusaSlicer #14784: Wipe tower blob crash mitigation

agoode, 2025-08-21:
### Description of the bug

Wipe tower blobs have been causing crashes and layer shifts for some time now:

Older posts:
- https://forum.prusa3d.com/forum/original-prusa-i3-mmu2s-mmu2-hardware-firmware-and-software-help/purge-tower-bumps-crashing-etc/
- https://www.reddit.com/r/FixMyPrint/comments/1ce2l3u/sloppy_wipe_tower_causes_collisions_and_print/

Newer:
- https://www.reddit.com/r/prusa3d/comments/1cuxy1d/new_mmu3_with_mk4_purge_tower_blobs/
- https://www.reddit.com/r/prusa3d/comments/1gf4sbu/wipe_tower_causing_collision_and_layer_shift/

I have been having a lot of trouble with some blobs recently that happen at different points in the print, even when reprinting the same gcode files, and I'd like to try to solve this with some changes to the wipe tower generation.

Some observations:
- Blobs are rare, but even one in a print can cause a crash and layer shift.
- Folks on forums suggest that changing parameters can help, but it's hard to know since blobs don't appear at the same time in each print.
- Blobs can appear just before or after an MMU filament change.
- When a misfeed happens, the MMU will usually retry (successfully!) but may end up making a blob. The rest of the system is tolerant to these kinds of transient issues except for the wipe tower implementation.
- When the nozzle is moving slowly during a wipe cycle and encounter a blob, the blobs do not interfere with nozzle movement, but can be converted into tall walls.
- Blobs that get pushed this way are several layers tall, so can cause problems for a long time.
- The nozzle can hit blobs and crash during the wipe cycle (during the fast moves), or during normal movement between parts.
- The tall walls are especially bad.

I'm also assuming that loads/unloads are going to be naturally nondeterministic, so no amount of dialing in of fixed parameters is going to solve the problem fully.

Some potential ideas:
- Since the nozzle doesn't crash when moving slowly, maybe we could do a kind of ironing step to make sure blobs are flattened before the nozzle leaves the wipe tower.
- We could route around the wipe tower during moves.
- We could move slowly over the wipe tower during moves.

I think some kind of ironing might be best. I don't know of other ways to handle the issue at gcode generation time, since blobs cannot be detected and reacted to during the print.

I'm attaching images of my latest print's wipe tower, which resulted in a pretty severe crash.

[image]

[image]

### Project file & How to reproduce

[blobbo.3mf.zip](https://github.com/user-attachments/files/21909706/blobbo.3mf.zip)

The project file is attached.

Filament 2: Prusament PLA Gravity Grey
Filament 5: Printed Solid Jessie PLA Blue Moon

Printer: CORE One with MMU3 (enclosed)

I've just cleaned and tuned the printer and MMU3.

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

macOS 15.6

### Printer model

CORE One with MMU3

agoode, 2025-08-21:
I hope to dig into the slicer code and try a few things, but it will be a little while before I can do that.

CappyT, 2025-09-03:
Following due to same issue

CappyT, 2025-09-18:
Strangely enough I have this problem ONLY when I use the HF nozzle (specifically HF hardened nozzle, but i think also the brass would do the same).

Swapping to non-HF makes perfect purge towers.

panfri787, 2026-02-19:
@Alienbliep Not working for me, same problem. Thanks anyway!

Mattamue, 2026-03-14:
This happens in my mk4s, without an enclosure

MrFlippant, 2026-08-18:
TLDR: Print this STL and use it to calibrate your nextruder filament sensor.
https://www.printables.com/model/1815141-nextruder-mmu-sensor-calibration-strip

Testing a theory.

Based on my experience with mmu on mk3 extruders, what I've just read on this page, and my understanding of how the nextruder filament sensing works with mmu, I would suggest the best course of action is to seek out some undersized filament. Use your calipers on every spool of cheap-AF filament you have, and try to find the smallest diameter you can. Then, use that piece of filament to recalibrate the sensor.

mk3 users could use the 1.5mm hex key for that purpose, but that can't be done with the nextruder, so you need to find filament, or maybe some wire, that is about 1.5mm diameter to use for calibration purposes

blobs happen because the sensor does not trigger properly when the filament is loaded in the gears, and so the filament is loaded too far without any extruder motion to spread out the unexpected extrusion. That's why putting in a negative loading distance helps, but does not always prevent blobs

the nextruder sensor design is clever and all, but there doesn't seem to be an easy way to ensure that the sensor triggers when slightly undersized filament is loaded

this could also be fixed by firmware/software making the extruder move for the entire loading process, but then the wipe tower will have to be larger to make sure it is not moving in the same area where extrusions may already exist.

you could probably print a length of filament, actually. extrude a hexagon that is 1.5mm point to point (inscribed polygon) and print that, then use that to calibrate the sensor.

I certainly could be wrong, but this is how I would tackle the problem at first.

BreakMate, 2026-08-19:
I wish that worked @MrFlippant. unfortunately, after doing the calibration that you mentioned (printed hexagon 1.5mm edge to edge) I still have serious layer shifts due to the blobs (I was printing at 0.1 layer height). 

it seems like the MMU3 only works well with a non-hf 0.4 nozzle and 0.2 layer height.

Alienbliep, 2026-08-19:
Did the MMU ever worked or is this problem only on CORE One?
The only thing that changed is it chamber and elevated temp. 

I think the problem lies in the fact the nozzle stops everytime on the same spot and burns into the filament while loading and unloading.

BreakMate, 2026-08-19:
@Alienbliep I have my mmu3 on a mk4s. It works well only at 0.4 nozzle and 0.2 layer height. everything else created blobs and layer shifts

zzqzzqzzq, 2026-08-19:
Its definitely worked for me when I had the MK4... (yes 0.4 nozzle, 0.2 layer height)

https://www.printables.com/make/281074

Maybe I got lucky..

I've had failures on the C1 even w/ the 0.4/0.2 combination.

Alienbliep, 2026-08-19:
Too bad. I dont know the solution too. The lift 1cm while unloading and loading gcode did not help either?

MrFlippant, 2026-08-19:
Well, poop. I guess I don't know enough about the Core One with MMU, then. Still, I feel like that's where the issue is, since it's such a random and rare (by filament change count) issue. 

Is anyone able to do the following?
1. Install a motor rotation visualizer of some kind, so we can see every turn of the motor. This reveals a LOT of what's happening during filament changes.
2. set up a camera so that the extruder can be clearly seen during the unload/load/purge sequence, including the visualizer
3. set up another camera on the screen showing sensor status during the entire print, assuming this can even be done on the buddy board like it can on the einsy, and synchronize that video with the other one.

I'm assuming not, as that would be a lot of trouble for possibly no gain, if everything appears to be working correctly and it still makes a blob. 

I just can't think of any other reason that blobs like people are showing can even happen. If the extruder isn't pushing filament out, leaving the nozzle over a single point for hours won't make a blob like those. You'd have to be printing water logged nylon or something for it to extrude that much without the motor actively pushing it out.

To make those blobs, the extruder needs to be stationary AND actively extruding. The only time it's stationary is when it believes there is no material present in the hot end. That can only happen during filament changing, before it knows that filament has been successfully loaded into the drive gears, and has not yet entered the melt zone. As soon as the sensor triggers, it loads the material into the melt zone, and then starts moving before it expects extrusion to occur. If the sensor has not triggered at the right time, then the filament could be extruding before motion begins again, causing the blob.

Yes, I'm just thinking out loud.

Alienbliep, 2026-08-19:
Thats an nice theory which i bet prusa uses too, but the problem is it stops in already deposited material which gets cooked. And always on the same spot.

MrFlippant, 2026-08-19:
In which case, the problem is the already deposited blob, isn't it? 
It's not the stopping in a blob that needs fixing, it's the creation of the blob it's stopping in. 
The pre-existing blob just happens to be in the place it stopped moving while going through the next filament change cycle.

Anyway, sorry my calibration idea didn't work, and for not being more helpful. Without being able to see the whole of the actions taking place during blob creation, it's hard to offer any other explanations. 

Hopefully you guys are able to reach a reliable solution. Sad that my MMU3 on my MK3 is more reliable. :\

Clete2, 2026-08-19:
From what I can tell the blobs happen right before unload during ramming. Changing ramming distance alleviated my blobs.

Alienbliep, 2026-08-20:
Maybe it pumps bubbles in it? 😅🤣

BreakMate, 2026-08-20:
There are 2 problems that happen with the purge tower. I like to call them Blips and Blobs. 

Blips happen during a ramming to unload the filament. the extruder goes back and forth 3 times to make the tip and little poops come out. they can stick up and get collected, then deposited somewhere else by the nozzle. This is usually not an issue if there was material underneath for it to stick to.

The Blobs are what seem to cause the layer shifts for me. they happen during a load and too much filament is released too soon and creates a wall relative to the layer that can exceed 1mm tall.

karlost, 2026-08-27:
I’m honestly getting really frustrated with this issue.

This is not some minor cosmetic bug — a blob on the wipe tower can cause the nozzle to crash into it, ruin a long multi-material print, cause layer shifts, and waste a significant amount of filament and time.

What is especially frustrating is how long this has been known without any clear indication that it is being actively addressed. For something this fundamental to reliable MMU printing, I would expect at least an official status update, a confirmed workaround, or some indication of whether a fix is planned.

Could someone from Prusa please clarify:

* Is this issue actively being worked on?
* Is there a known reliable workaround?
* Is a proper fix planned for an upcoming PrusaSlicer release?

MMU reliability depends heavily on the wipe tower working correctly. Please don’t let such a critical issue sit unresolved without communication.

zzqzzqzzq, 2026-08-27:
- <cynically>
- Is this issue actively being worked on?
- No.
- Is there a known reliable workaround?
- Buy Indx?
- Is a proper fix planned for an upcoming PrusaSlicer release?
- Indx support is already in PrusaSlicer.  Nothing to do.
- </cynically>

While there's a great deal of cynicism, and tongue in cheek, its not entirely tongue in cheek.   The year having passed with no updates doesn't give me (or I suspect anyone else) a wam fuzzy.

Alienbliep, 2026-08-27:
No it is as abandoned as the titanic.
Indx came and all r&d people abandoned this system.
I was and am still waiting for an top hat for the core one L. So do a lot of people with less money that want to transfer their mmu3 to the core one and not buy indx.

Very frustrating because i still love the system even more with the indx and its problems. It just does not print as cleanly and reliably at times as the nextruder.

I only use the mmu for choosing a spool at the moment. Not really for color prints. It really sucks.

Still i dont understand why mmu3 on mk4s was good and on the core one this happened. That can only be because of elevated temps.

Hell, they could even come up with an purge bucket with wiper because that could be invented for the indx in weeks. But that requires some designing and coding.
Dont know if there is an place where and bucket and wiper would fit with the nextruder not being mounted high up.

karlost, 2026-08-27:
At this point, this honestly feels like a slap in the face to existing customers.

Prusa is **still selling the MMU3**, while a serious issue that can ruin an entire print has been sitting here for roughly a year without a proper fix or even a clear roadmap.

And the answer is basically: *buy INDX?*

For an ordinary customer, INDX is hardly an answer. Apart from the additional cost, the availability itself has become a joke — as somebody put it elsewhere: **“order in the morning, get it next year.”**

What frustrates me most is that this is not some obscure cosmetic bug. A blob on the wipe tower can cause a nozzle crash, layer shift and destroy a many-hours-long print. Yet customers are left experimenting with loading distances, sensor calibration, nozzle combinations and community workarounds while the product continues to be sold.

It reminds me of companies making strategic decisions while ignoring what part of their existing customer base actually wants. Sony is now moving PlayStation toward digital-only releases and abandoning physical discs for new games from 2028. Nokia famously failed to react quickly enough when the smartphone market changed around it.

Different situations, obviously, but the lesson is similar: **being convinced that customers will simply follow your new direction is dangerous.**

I like Prusa products, which is exactly why this is so frustrating. But situations like this increasingly make me look at the competition. You cannot build customer loyalty by selling hardware and then effectively abandoning a major problem when the next shiny product arrives.

And in 2026, with the development tools available today — including AI-assisted development — watching a software-related problem this significant sit unresolved for a year is very difficult to understand.

INDX may be the future, but it should not become an excuse to stop properly supporting products that Prusa is still actively selling.

Alienbliep, 2026-09-02:
I dont think Prusaslicer 3 will fix this.
I have ordered indx to leave the mmu system.

BreakMate, 2026-09-02:
There is going to be a special place next to my INDX for the MMU3 after I detach it from my MK4S

Alienbliep, 2026-09-02:
Yeah i think it still needs to be solved.

Clete2, 2026-09-02:
I really like the MMU3 and would love an official fix.

BoiboSAn, 2026-09-11:
I have this issue to, makes me hesitentant to start prints with many swaps as the ods of a failed print is to large. MMU is just month old.

As other mentioned the issue is that the head parks on the purge tower, really close to the tower (touching even), causing a blob and strands working itself loose. Purge towers look like messes.

My first test print with the mmu failed due to this, got a layer shift... Thats sad really for a quite expensive product.

As it is now I might have to demand a fix or money back.

zzqzzqzzq, 2026-09-18:
Till someone confirms this is actually *fixed* in 3.0, it is highly likely NOT a legacy issue, despite the labeling.

ravenlp, 2026-09-24:
I just had this issue with 3 models sliced with PrusaSlicer 3.0.0-alpha11 and 3.0.0-alpha12, so seems to be an ougoing issue still.
Trying some of the workarounds mentioned in this thread right now.

davidnogol, 2026-09-29:
I have same issue on MK4S + MMU3 (without enclosure), latest firmware on printer and MMU. Tried several tips in changing settings in slicer, but I had no success. I have MMU3 only few days, I printed 3 smaller prints successfully. Now I want to print larger objects with a lot of filament changes, but after 1-2h of prints blobs accumulates and layer shifts. Its really frustrating, I lost a lot of filament and testing take a lot of time. Missing some official troubleshooting guide for this issue from Prusa :(

[image]

krcollier, 2026-09-30:
Has anyone tried adjusting the ramming settings? Lowering the volumetric speed and/or increasing the total ramming time?

It looks like the blobs I get are caused at/during the ramming sections of the wipe tower. I'm thinking maybe the filament I'm using can't extrude or adhere at these settings. 

Going to risk this 1.5 day print before changing settings. I'm already halfway through and haven't had a layer shift, even though a blob has formed on one color on the wipe tower. That specific color is from Kexcelled with a lower volumetric extrusion rate, about 2/3rds the other Sunlu Colors.

ravenlp, 2026-09-30:
I had good results (meaning I was able to end a day long print without any assistance) using this setting changes to the originals ones:

`Extra loading distance: -52mm` on the machine settings
and 
`Wipe tower purge lines distance: 200%` on the profile settings

Mawos, 2026-10-06:
> Till someone confirms this is actually _fixed_ in 3.0, it is highly likely NOT a legacy issue, despite the labeling.

I am seeing this issue in PrusaSlicer 3.0, just downloaded it this week. Very frustrating. I’ve been working through various suggestions but I’m still seeing several blobs form

archiebug, 2026-11-14:
I have this issue too as it is causing layer shifts