prusa3d/PrusaSlicer #14785: Object does not fit on the print bed

AudaxSoftware, 2025-08-21:
### Description of the bug

An object with the maximum depth of the print bed (here Core One 220 mm) will not fit on the print bed, although it should fit. Upside down, it fits.

### Project file & How to reproduce

[Archiv.zip](https://github.com/user-attachments/files/21919150/Archiv.zip)

[image]

[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.2

### Operating system

macOS Sequoia 15.6.1

### Printer model

Prusa Core One

AudaxSoftware, 2025-08-21:
In principle, you are right. However, the dimensions given in the window say exactly 220 mm. And 220 mm must fit. If it was 220.0001, then it wouldn't fit.

u89djt, 2025-08-21:
You're seeing a maximum of two decimal places there. The width is less than 0.01 too big. If you unlock the aspect ratio and type 219 in the box you get this:
[image]
Then type 220 and you get this:
[image]
The resulting object is different to the one you provided, but the difference is too small to see in the numerical display.
That's why it's happening. Your object in the original file is wider than 220mm, but not as much as 220.01mm.
You could ask for a feature to increase the precision of numerical displays?

closed 2026-09-29 by prusaslicer-bot[bot]