prusa3d/PrusaSlicer #14787: prusaslicer using wrong graphics driver

ptschj, 2025-08-22:
### Description of the bug

i am using prusaslicer 2.9.2 in a virtual machine. this has been working file for a few years, until i recently upgraded the display protocol and graphics driver. now, for some reason, prusaslicer is using cpu display, and the performance is abysmal. just trying to rotate the model goes down to about 2 seconds per frame. and i don't mean to frames per second, just to be clear. i attached a system info below. as you can see, for some reason it's using display vendor of vmware instead of nvidia. the vm has a discreet nvidia gpu passed to it, which is working fine for literally every other app on the system. how can one fix this. copying the opengl32.ddl file into the main installation folder doesn't help either. using the portable version doesn't either. the problem persists when using the gcode viewer. 

`PrusaSlicer
Version:   2.9.2
Build:     PrusaSlicer-2.9.2

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (build 22631), 64-bit edition
Total RAM size [MB]: 34,353MB
OpenGL installation
GL version:   3.3.0 (3.3 (Core Profile) Mesa 12.0.0-rc2)
Profile:      Core
Vendor:       VMware, Inc.
Renderer:     Gallium 0.4 on llvmpipe (LLVM 3.6, 256 bits)
GLSL version: 3.30.0
Textures compression:       Disabled
Mutisampling: Disabled
<details>
<summary>Installed extensions:</summary>
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clip_control
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robustness
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shader_stencil_export
GL_ARB_shader_subroutine
GL_ARB_shader_texture_lod
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map_array
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ATI_blend_equation_separate
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_blend_equation_separate
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_sRGB
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_pixel_buffer_object
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_shader_integer_mix
GL_EXT_texture_array
GL_EXT_texture_compression_rgtc
GL_EXT_texture_integer
GL_EXT_texture_mirror_clamp
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_multimode_draw_arrays
GL_KHR_context_flush_control
GL_KHR_debug
GL_MESA_pack_invert
GL_MESA_texture_signed_rgba
GL_MESA_ycbcr_texture
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_packed_depth_stencil
GL_OES_EGL_image
GL_OES_read_format
</details>
`

### Project file & How to reproduce

to reproduce, using prusaslicer in omnissa horizon 8.16 with a discreet gpu

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11, 23H2

### Printer model

MK3S+

mikebad1984, 2025-09-22:
I have this same issue while over RDP and decided to go down the rabbit hole to see if I could fix it. 
I found the OpenGL start code around slic3r/GUI/OpenGLManager.cpp and tried to dig deeper but if I'm interpreting everything correctly it looks like Windows+RDP just doesn't play nice with OpenGL.
It didn't work for me but in case it works for you, one thing I tried is enabling hardware emulation by using Group Policy Editor (gpedit.msc) and going to
Adminstrative Templates->Windows Components->Remote Desktop Services->Remote Desktop Session Host->Remote Session Environment then enable "Use hardware graphics adapters for all Remote Desktop Service sessions"
(Be sure to reboot)

ptschj, 2026-03-14:
> I have this same issue while over RDP and decided to go down the rabbit hole to see if I could fix it. I found the OpenGL start code around slic3r/GUI/OpenGLManager.cpp and tried to dig deeper but [...]

hast thou made any progress since?

JesseMigdal, 2026-03-15:
No, I gave up shortly afterward messing with RDP.  I think I had some luck
using Chrome Report desktop , but I honestly got so fed up that I ended up
just making the files available to me over the network and working locally.

ptschj, 2026-04-01:
> No, I gave up shortly afterward messing with RDP.  I think I had some luck
> using Chrome Report desktop , but I honestly got so fed up that I ended up
> just making the files available to me over the network and working locally.
> […](#)

it seems to me my issue is the horizon display protocol my vm uses not playing nicely with opengl that prusa is using. who knows when that's gonna get solved though....

closed 2026-09-29 by prusaslicer-bot[bot]