prusa3d/PrusaSlicer #14800: /mnt directory not visible in file operations

VikOlliver, 2025-08-28:
### Description of the bug

When trying to export any G code, the /mnt directory is not visible in PrusaSlicer's file manager window.

As the /mnt directory contains the mounted USB drive, which is user-accessible (works fine in PrusaSlicer 2.7.4), I expected to navigate to there and save my bgcode on the USB drive.

What happens is that the left sidebar of the PrusaSlicer file manager does not contain the shortcut to the mounted USB drive that I use successfully in V2.7.4.  The "+ Other Locations" option has a directory entry called "Computer" and clicking on that reveals the root directory contents - except for "mnt". There is also an extra directory called "app" listed which does not exist in my root file system.

### Project file & How to reproduce

[no_mnt_dir.zip](https://github.com/user-attachments/files/22021321/no_mnt_dir.zip)

[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.2

### Operating system

Ubuntu 22.04.5 LTS, installed as flatpak from prusa3d.com

### Printer model

5-Head XL

Jan-Soustruznik, 2025-08-28:
Hello, @VikOlliver , yes this is know issue, please ise the Flatseal to add the `/mnt directory`

> Hello, as you correctly determined, this is a flatpak permissions issue. Personally I would recommend [flatseal](https://flathub.org/apps/com.github.tchx84.Flatseal) to permanently add paths you want.
> 
> [image]

source: https://github.com/prusa3d/PrusaSlicer/issues/13820#issuecomment-2563593643

Closed as duplicate of #13820

closed 2025-08-28 by Jan-Soustruznik

VikOlliver, 2025-08-28:
FYI I fixed the issue without installing another 3rd party package using:

`flatpak override --user --filesystem=/mnt`

on the command line.