prusa3d/PrusaSlicer #14802: Print Quality Issues on Small Angle Surfaces (10°) Caused by Infill Handling

freakydude, 2025-08-29:
### Description of the bug

**Description**

On surfaces with small angles (e.g. 10°), PrusaSlicer 2.9.2 produces significant print quality issues due to the way unsupported layers above infill/support structures are generated.

Instead of bridging across gaps, PrusaSlicer prints short infill lines "in the air," reversing direction mid-air. This causes the filament to curl up and down, resulting in blobs, ripples, and surface artifacts on the following solid/perimeter layers.

**Steps to Reproduce**

- Slice a model with a ~10° angled surface.
- Use PrusaSlicer 2.9.2 with default settings (see attached images).
- Observe how the slicer places infill lines not anchored to surrounding geometry.
- Print the model and compare surface quality.

**Actual Behavior (PrusaSlicer 2.9.2)**

- Infill lines are generated unattached to the structure.
- Print head changes direction mid-air. 
- Filament curls → causes bulges and visible artifacts 2 layers above.

[image]

**Print**

[image]

**Expected Behavior (Reference: OrcaSlicer 2.3.1-alpha)**

- Infill gaps should be treated as bridges.
- Unsupported regions should be printed as continuous bridged perimeters.
- No mid-air retractions or direction changes.
- Result: clean surface without curling artifacts.

[image]

**Print**

[image]

**Environment**

- PrusaSlicer version: 2.9.2
- Reference (works correctly): OrcaSlicer 2.3.1-alpha
- Printer/Settings: Same printer and (as close as possible) same settings across slicers

**Summary**

PrusaSlicer currently generates unsupported infill in a way that introduces major print defects on shallow-angle surfaces. Treating these layers as bridges (as OrcaSlicer does) would prevent curling and result in cleaner prints.

### Project file & How to reproduce

[bugreport-stand-prusa.zip](https://github.com/user-attachments/files/22042081/bugreport-stand-prusa.zip)
[bugreport-stand-orca.zip](https://github.com/user-attachments/files/22042080/bugreport-stand-orca.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Prusa 2.9.2

### Operating system

Windows 11 24H2 26100.4946

### Printer model

Voron 2.4 with SB

u89djt, 2025-08-29:
(community triage)
Both slicers switch between regimes - bridging loops to zigzag - when a decision is made not to include top solid infill, and that decision happens at a different threshhold somewhere. It might be a bug, but it might be a decision that places the threshhold differently.
You can see this if you reduce the number of perimeters in the prusa slice to 2 from 3:
[image]
And increase the number of loops in orca from 3 to 4:
[image]
How does your surface quality look if you increase the loop count in Orca to have it switch regimes?

u89djt, 2025-08-29:
Getting on with perturbing the system to find triggers: if you up the infill/perimeter overlap to 27%, it switches to the preferred regime in your specific example for prusaslicer. Not something you'd want to be forced to do, but it might be revealing. Or even useful to you right now, I don't know. 
[27pc-overlap-hack-bugreport-stand-prusa.zip](https://github.com/user-attachments/files/22050689/27pc-overlap-hack-bugreport-stand-prusa.zip)
[image]

u89djt, 2025-08-30:
Last input from me, just to capture the issue in a standard profile and basic object:
[10degreeslope-3perimeters-thicknessregime.zip](https://github.com/user-attachments/files/22057147/10degreeslope-3perimeters-thicknessregime.zip)
[image]

freakydude, 2026-09-02:
Hi,

sorry I have to say that, but the issue still exists. 

Started with a clean PrusaSlicer 3.0.0-alpha11 config, 
selected the Prusa i3 mk3s printer, 0.4mm nozzle and the prusament pla default settings

Re-sliced my old model from the original issue. 

Result: Same In-Air print like before (picture is taken, cutting of the bottom slices this time). If you print this, the infill is ends in air and will change direction in air with results into print failures at this regions

[image]

Best regards
freakyDude