prusa3d/PrusaSlicer #14808: Early spin-up of cooling fan for improved overhang/bridge printing performance

Mike-FUT, 2025-09-02:
**Is your feature request related to a problem? Please describe.**
When printing overhangs and bridges with my MK4S the surface quality is in many cases not good enough so the measurements and optics suffer. I noticed in moving direction of the head ther beginnings usually look worse than the endings of overhangs.

**Describe the solution you'd like**
Constant surface quality of short/long bridges and overhangs

**Describe how it would work**
Spin up the filament cooling fan early enough so it allows cooling a bridge/overhang when it reaches the starting point.
Currently the fan starts spinning up directly at the beginning of a bridge for example, which is far too late and sometimes the fan doesn't even reach the set speed before reaching the end of the bridge.

**Possible technical solution**
For testing purposes I wrote a (very messy, very simplified) G-Code post-processor.
What it does:
1. it analyzes all G-Code operations that takes time (travel moves, extruding, retraction, etc.) and adds a absolute timestamp in microseconds behind these lines as comment
2. it checks for fan speeds commands that are higher than the previous fan speed value
3. if an increased fan speed command is found it calculates the time it takes between the smalles fan speed command during the last 1.2 seconds and the requested fan speed (using simplified linear interpolation, assuming that my MK4S fan takes 1.2 seconds from 0% to 100%).
4. it looks for the latest line just before the the fan speed increase command that contains a timestamp.
5. it uses the time to spin up the fan to the required speed (calculated in 3.) and substracts it from the timestamp in 4.
6. it searches for a timestam equal or smaller than the calculated value in 5. and inserts the fan speed increase command just before that line.
7. it comments all fan speed change commands between the line from 6. and 2.

So to summarize: The G-Code postprocessor will make sure, the increased fan speed is reached when the nozzle is at the starting point of a bridge or overhang.
While the code from my postprocessor is very messy (I'm no programmer) the results of the printed items is significantly improved.

If that is all to complicated to properly implement in PrusaSlicer another, very simplified option could be the Bambu Studio way of dealing with the spin-up issue: They add a user configurable, fixed time where the spin-up commands are pulled ahead. While this might lead to unnecessary high cooling it's still better than the current PrusaSlicer way where the cooling is simply too late.

Mike-FUT, 2025-09-02:
@neophyl, thank you, seems I'm not good in finding things here at Github.
But since this feature request seems to be out there for five years now I might probably clean up my messy post-processor code (where I described the logic above) and at least put that in the Prusa forum so others can use it as it brings clearly noticable print advantages for me.

SuperGripM, 2025-12-14:
This makes a lot of sense. Especially for users with large blower fans that take a long time to spin up.

closed 2026-09-29 by prusaslicer-bot[bot]