prusa3d/PrusaSlicer #14811: PrusaSlicer Linux cannot access /mnt/

CDRXavier, 2025-09-03:
### Description of the bug

I have my linux set up so my Documents folder is a symbolink to folders in another partition. Prusaslicer is unable to follow that, which is fine. However it is also unable to access the /mnt/ folder where the partitions are mounted in.

It is able to access /media/ for SD cards.

Is it privilege thing? But it dont make sense, as other user-space snap applications are able to (access suuch locations), such as firefox.

### Project file & How to reproduce

Try to access the /mnt/ folder, you cannot. And if you try to drag and drop other files (located under the /mnt/ folder) you will get
"Prusaslicer has encountered a error"
`basic_string::erase: __pos (which is 18446744073709551615) > this->size() (which is 0)`

If you move the file to somewhere accessible (e.g., desktop), everything is fine.

Attached is the project file. There is nothing wrong with that, since it is a directory issue.
[1366ilm-Chamfer_3mf.zip](https://github.com/user-attachments/files/22107167/1366ilm-Chamfer_3mf.zip)

[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.2

### Operating system

Ubuntu 24.04.02 LTS

### Printer model

Prusa i3 MK3S/+

neophyl, 2025-09-03:
Please search for both open and closed issues before opening new ones.  For example this was from a few days ago https://github.com/prusa3d/PrusaSlicer/issues/14800  exactly the same issue reported which itself is a duplicate of many others.

Jan-Soustruznik, 2025-09-03:
Closed as dulicate of #13820

source: https://help.prusa3d.com/article/install-prusaslicer_1903 
[image]

closed 2025-09-03 by Jan-Soustruznik

CDRXavier, 2025-09-07:
> Please search for both open and closed issues before opening new ones.

Oh. Well, I didnt search closed ones. I suppose that does mean something.

> source:

Apparently this section of the "how to install" is not very visible as I scrolled past it without noticing. Thanks.