prusa3d/PrusaSlicer #14817: Organic Support Crashes when select Extruder 3 (works for default extruder) 2.9.2 and 2.9.3b1

tba-gh, 2025-09-04:
### Description of the bug

Slicing crashes or stops at 70% when selecting Extruder 3, works fine with default Extruder. XL5T.

Project file attached.

[death_organic_crash.zip](https://github.com/user-attachments/files/22161896/death_organic_crash.zip)

### Project file & How to reproduce

Slicing crashes or stops at 70% when Extruder 3 selected. Leaving for extended time does not resolve. Works fine with default Extruder and completes the slice in less than a minute. XL5T. Repeatable with both latest 2.9.2 and 2.9.3b1.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11 build 26100

### Printer model

XL5T

u89djt, 2025-09-05:
(community triage) Slicing doesn't crash for me in 2.9.2 under win11 if I change to extruder 3 (or 1,2,4 or 5) in your project like this:
[ext3_death_project.zip](https://github.com/user-attachments/files/22169284/ext3_death_project.zip)
[image]
Someone with the beta installed might check.
Is that the change you mean? Cheers!

u89djt, 2025-09-05:
Given that the crash isn't reproduced on a different system in one of the versions you list, we know that the specifics of yours are relevant. Have another look at the bug submission guidelines and go ahead and post all the details they describe.
It looks like this:

PrusaSlicer
Version:   2.9.2
Build:     PrusaSlicer-2.9.2

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (build 26100), 64-bit edition
Total RAM size [MB]: 33,435MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 576.52)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 4090/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 330 lines cut ...]
GL_NV_transform_feedback2
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

tba-gh, 2025-09-05:
Thank you for looking into this. The thing that I don't understand is why switching a head would behave any differently regardless of my processor or BIOS. It slices fine when I select the DEFAULT TOOL head and then doesn't complete the slice when I switch to TOOL 3. How can that be anything to do with my machine? Are there some logs I can provide from Prusa Slicer?

PrusaSlicer
Version:   2.9.2
Build:     PrusaSlicer-2.9.2

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (build 26100), 64-bit edition
Total RAM size [MB]: 34,201MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 572.16)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3080/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
[... 329 lines cut ...]
GL_NV_transform_feedback2
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

u89djt, 2025-09-05:
The processor issues are deeply complex and subtle - they can hit stuff you wouldn't have any way of predicting. Given that rebooting changed the way the slicer behaved, we know there's something counter-intuitive happening. Have a look for updates relating to your specific processor and/or motheboard. If you find anything, you could report that in a comment, and maybe edit into your report at the top there, so someone else with a similar setup has the opportunity to find it.