prusa3d/PrusaSlicer #14819: Full Ukrainian translation for 2.9.2

DrFOXUA, 2025-09-05:
[Full Ukrainian translation for 2.9.2](https://github.com/DrFOXUA/PrusaSlicer-2.9.2_UK.git)

YuSanka, 2025-09-08:
Це просто чудово :) Раді, що будемо мати переклад і до української мови. Зробіть будь ласка PR і ми з радістю його приймемо :) Але зверніть увагу на те, що лише .МО файлу недостатньо. PR має містити і .РО файл ;)

That’s just wonderful :) We’re glad that we’ll have a translation into Ukrainian as well. Please create a PR and we’ll be happy to accept it :) But note that providing only the .MO file isn’t enough — the PR must also include the .PO file ;)

DrFOXUA, 2025-09-08:
Я не очень понял что такое PR.
Мой файл РО сильно отличается от оригинала, но я выложу его рядом с МО файлом.
(извините за одесский язык)

YuSanka, 2025-09-09:
Thanks for your contribution. We manually merged your PO file with [commit](https://github.com/prusa3d/PrusaSlicer/commit/b9e33f4b096284903959057379ef48857f429ea3).

closed 2025-09-09 by YuSanka

DrFOXUA, 2025-09-09:
[resources/localization/uk/PrusaSlicer_uk.po](https://github.com/prusa3d/PrusaSlicer/blob/b9e33f4b096284903959057379ef48857f429ea3/resources/localization/uk/PrusaSlicer_uk.po)
это не совсем мой файл.
Может я чего-то не понимаю, но зачем добавлять в РО файл ненужные строки, которых нет в РОТ файле, из 1800 строк добавленых из РО файла в папке wx_locale будет использовано ну может строк 20. Остальные ЗАЧЕМ???
И еще. 
В моем файле есть коменты касающиеся непереведенных строк, которые желательно обозначить в исходных файлах...

DrFOXUA, 2025-09-09:
Добавил **Полный** перевод для версии 2.9.3-rc1