prusa3d/PrusaSlicer #14821: Variable layer height removes support on other side of part

drott, 2025-09-06:
### Description of the bug

When slicing a hook shaped part with two tubes, and using variable layer height, after the first range of denser layers ends, there is no support in the other side of the part. 

### Project file & How to reproduce

[Hook tighter print.3mf.zip](https://github.com/user-attachments/files/22189626/Hook.tighter.print.3mf.zip)

Slice, scroll through layers.  Observe that above lower hook end no supports are generated on the other, long side, of the part.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.3 beta1

### Operating system

Mac

### Printer model

Prusa Core One

drott, 2025-09-06:
[image]

drott, 2025-09-08:
You mean the infill density is so wide that here the inner shape/square between infill lines would is as the wide as the part? Might be, in that case closing as working as intended.

u89djt, 2025-09-08:
Your suggestion makes more sense.
To open it up a bit more, here's the model stretched along that leg to show a few more transitions. Infill anchor length is set to zero so we only see the intended infill pattern.
[longer var and const Hook.zip](https://github.com/user-attachments/files/22203706/longer.var.and.const.Hook.zip)
[image]
[image]
It makes more sense to be a minimum extrusion length than where the sheets cross. I think my suggestion was wrong.