prusa3d/PrusaSlicer #14828: Typind in help box problem

Boom23-Prusa, 2025-09-09:
### Description of the bug

When I select and type a word in the Search field, the characters do not show . If I press Enter after that, the Filament Options Abrasive Material is highlighted. When I try to get out of the Search, the Search List blinks. I need to press Esc few times to get out of Search.

This happens on my Windows 11 ROG Zephyrus G14 (AMD) and my Dell laptop (i7). It does not happen on my Dell desktop (i9) computer. I am using PrusaSlicer 2.9.3 RC1 but I had the same problem since 2.9.2, which is when I started using PrusaSlicer.

### Project file & How to reproduce

Open PrusaSlicer 2.9.3 RC1, click on the empty Search field, located next to "Prusa Connect". 
Type anything. The characters do not show.

Note that if I click on the actual Search icon, in the same field, the Search works as it should.

[image]

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.9.3 RC1

### Operating system

Windows 11

### Printer model

Prusa Core One

EastArctica, 2025-09-30:
Also experiencing this on my Windows 11 (24H2 26100.6584) Surface Pro 9, but not on my Windows 11 (Also 24H2 26100.6584) Desktop (both intel).

Perhaps this is a general laptop issue?

Boom23-Prusa, 2025-09-30:
I have the latest Windows 11 (26100.6725) with Slicer Version 2.9.3 and the problem still exists on my laptop.

closed 2026-09-29 by prusaslicer-bot[bot]