prusa3d/PrusaSlicer #14831: I cannot open files from external drives in Linux Mint 2x.x and PrusaSlicer upper than 2.9

victornavarrofuster, 2025-09-10:
### Description of the bug

In versions 2.9.x installed from Flatpak, stl files from secondary drives cannot be opened, as the location path /mnt does not appear. Shortcuts cannot be added to the ‘Other locations’ bar either. Attached is a screenshot of version 2.9.2 and version 2.8.0, which is the one I am using and works perfectly.

Thank you for your assistance.

[image]
[image]

### Project file & How to reproduce

In versions 2.9 for Linux installed from Flatpak, secondary disk paths cannot be opened. The Linux Mint /mnt folder does not appear.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.9.2

### Operating system

Linux Mint 22.2

### Printer model

Anet A8

neophyl, 2025-09-10:
They need to re-pin this back at the top again.  There have been so many duplicates of this.
Please search for existing issues, both open AND closed.

Its a flatpack issue. https://github.com/prusa3d/PrusaSlicer/issues/13820#issuecomment-2563593643

victornavarrofuster, 2025-09-10:
Thanks neophyl!!!. I have solve it!! 

"I've managed to fix mine by...   

flatpak override --user --filesystem=host com.prusa3d.PrusaSlicer

To give the app access to all filesystems."

https://github.com/prusa3d/PrusaSlicer/issues/13820#issuecomment-2573614106