prusa3d/PrusaSlicer #14839: Excessive tool moves after nozzle wipe

Kolineal, 2025-09-13:
### Description of the bug

When slicing a multimaterial model for Prusa XL after the bed is levelled end tool nr 2 purged, tool 1 does the purging but instead of going back to park as normal, it does a zigzag motion over the whole bed. It looks like the points it goes to, align with the 2 opposite corners of each instance of the model on the bed.
Apart from being unnecessary time waste, the extruder oozes plastic all over the bed , which ruins the appearance of all parts

[image]

### Project file & How to reproduce

[zigzag issue.3mf.zip](https://github.com/user-attachments/files/22309947/zigzag.issue.3mf.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.2

### Operating system

macOS Sequoia 15.5

### Printer model

XL 2 head

u89djt, 2025-09-13:
Many hands make light work :) I'll have a look at the temperature thing next time I'm here if nobody else has picked it up.

u89djt, 2025-09-13:
BTW, you could do this:
[21x zigzag issue.zip](https://github.com/user-attachments/files/22311128/21x.zigzag.issue.zip)
That would also halve the slowdown per item due to that temperature issue :)

closed 2026-09-29 by prusaslicer-bot[bot]