prusa3d/PrusaSlicer #14842: Extruder fan speed setting

cedrickevin936-eng, 2025-09-14:
**Is your feature request related to a problem? Please describe.**
If I print filaments with a high nozzle and chamber temperature, the heat tends to creep up to the extruder gear, softening the filament to early making it unable to extrude. I know that the extruder calling fan is not running at full speed so being able to adjust its power would be beneficial. 

**Describe the solution you'd like**
A similar setting to the part cooling fan where you can input a % fro the speed of the extruder fan out of 100

**Describe how it would work**
It would cool the midsection of the nozzle more, stopping the heat to reach the extruder gear

**Describe alternatives you've considered**
I've limited the chamber temp which helps, but causes more warping and decreases material properties/layer adhesion

**Additional context**
Add any other context or paint drawing about the feature request here.

pyro9, 2025-09-14:
Surely the extruder/hotend cooling fan should be 100%?
On most 3D printers (all that I have seen), it's not even controllable.

u89djt, 2025-09-14:
(from another user taking an interest) 
We don't know what printer is being used, or what temperatures are involved. Possibilities include:
- Chamber temperature is too high for it to be physically possible to cool the heatsink sufficiently with the airflow the installed fan can achieve
- The fan is on backwards
https://help.prusa3d.com/article/extrusion-stopped-mid-print-heat-creep_1948

pyro9, 2025-09-14:
@u89djt It could just be a worn out fan. I had one that had gotten slower bit by bit so I didn't really notice until I had heat creep issues and compared to a new fan.
But that's not something the slicer can address.

rtyr, 2025-09-15:
Hotend/extruder/heatbreak fan PWM should be handled by the firmware, which is basically always the case. Frankly I am not aware of any printer which will let you control heatbreak fan PWM directly and it would be not wise, since it can lead to printer damage in some cases. Some Prusa3D printers can change heatbreak fan PWM based on target heatbreak temperature.

Anyway, I am afraid this task is not for slicer to solve.

closed 2025-09-15 by rtyr