prusa3d/PrusaSlicer #14863: Pre-print purging issues

Ozzy-Artist, 2025-09-17:
### Description of the bug

Hi there, Prusaslicer 2.9.3 is doing weird stuff. On my XL After tool #1 completes bed levelling it tries to purge even though it's not used or activated to print in the slice. Then, the first tool that is to print (#5 in this case) is selected, goes to the purge area at the front of the plate... and sits there to heat up, doesn't purge and then goes right to the first layer print.
I reinstalled Prusaslicer and am running the XL firmware version 6.2.6+8948 but the problem persists. So, it's definitely a slicer issue as it happened after I updated to 2.9.3 and before (and since) I updated the FW  to the current version.  I cannot share files here as the model is proprietary so I'm attaching a test model slice with identical settings, profiles, erc.

[Ozzy's_Slicer_Issue.zip](https://github.com/user-attachments/files/22393281/Ozzy.s_Slicer_Issue.zip)

### Project file & How to reproduce

[Ozzy TEST.zip](https://github.com/user-attachments/files/22393297/Ozzy.TEST.zip)

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.3

### Operating system

Windows 11

### Printer model

Prusa XL 5 Toll head

rtyr, 2025-09-18:
I looked at your project, and I can see you are using incorrect profiles.

Your custom printer profile is based on profile for single-tool XL (on very old profile), where you probably just increased number of extruders, which will not work properly.

You have to enable and use 5TOOL XL in the configuration wizard, it uses different start gcode, specific toolchange commands etc.

Feel free to contact our tech-support at www.prusa3d.com for assistance.

[image]

closed 2025-09-18 by rtyr

Ozzy-Artist, 2025-09-18:
I checked the file I sent you and the file I’m printing  and your conclusion is wrong! Both are set to print with the 5 TOOL XL Input Shaper 0.4 nozzle. I have created a new post on Prusa3D GitHub and included the file again for proper analysis. CheersOzzy

xl5t.jpg (view on web)