prusa3d/PrusaSlicer #14865: Feature request: Please update Linux build docs. <--Successful Build Steps Here.

themanyone, 2025-09-18:
**Is your feature request related to a problem? Please describe.**
Build instructions haven't worked out of the box in years.
If we want to attract new developers, we need to make this straightforward.

**Describe the solution you'd like**
Update docs.

**Describe how it would work**
Make build process fast and trouble-free.

**Describe alternatives you've considered**
Searching the web and StackOverflow, Trying unsuccessfully to build unnecessary and outdated deps.

**Additional context**
Comments?

## Preparation

Get fresh snapshot of your branch for this build test.

```
# If in a hurry, this might work
git clone --depth 1 --branch [my_branch]  [my_project]

# If you run into problems:
# fetch remaining commits for git log, git bisect, etc.
git fetch --unshallow

# speed up cmake
export CMAKE_BUILD_PARALLEL_LEVEL=$(nproc)

# Suppress CMake Errors when builds mention earlier version of cmake
export CMAKE_POLICY_VERSION_MINIMUM=3.5

# CMakeCache.txt trips you up if you try to build the same project again.
rm CMakeCache.txt #fixes error with missing LibBGCode & GLEW_LIBRARIES, etc.
# If that fails, just wipe out the build directory and start fresh.

```

## Build deps

We need to update build docs. For one, you do not need to build all the deps.
Simply remove the +Folders you don't need.

```
cd deps
# Remove deps for which native -devel packages are available
# You can always `git checkout` these +Folders later, if ever needed.

# [Fedora's Imath-devel is bugged](https://bugzilla.redhat.com/show_bug.cgi?id=2395651).
# So, we have installed `Imath` from git (defaults to /usr/local/).
# Thus, we keep OpenVDB and GLEW because they depend on `Imath`.
# These are configured to build static, so need to do static build of Prusaslicer.
rm -rf +Boost +TBB +GMP +CURL +Eigen +OpenSSL +Qhull

# We will now make sure we have system libs for the deps we deleted.
sudo dnf install boost-devel tbb-devel gmp-devel libcurl-devel eigen3-devel openssl-devel qhull-devel 

# Update PKG_CONFIG_PATH so it picks up /usr/local (should probably add to .bashrc)
export PKG_CONFIG_PATH="/usr/local/lib64/pkgconfig:/usr/lib64/pkgconfig:$HOME/.local/lib/pkgconfig:/usr/share/pkgconfig"

rm -rf build
mkdir build
cd build
cmake .. -DDEP_WX_GTK3=ON
make # -j$(nproc)
```

# Build PrusaSlicer
```
#change directory to PrusaSlicer/
rm -rf build
mkdir build
cd build

# Sometimes we want to install dependencies in /usr/local
# Check & update PKG_CONFIG_PATH (should probably be in .bashrc)
# Prefer /usr/local libs, followed by system libs
export PKG_CONFIG_PATH="/usr/local/lib64/pkgconfig:/usr/lib64/pkgconfig:$HOME/.local/lib/pkgconfig:/usr/share/pkgconfig"
export CMAKE_POLICY_VERSION_MINIMUM=3.5
rm CMakeCache.txt

# Build static because deps are configured for static build and we're lazy
# Point CMAKE_INSTALL_PREFIX to $HOME/.local for single user debug install
cmake .. -DSLIC3R_STATIC=ON -DSLIC3R_GTK=3 -DSLIC3R_PCH=OFF \
    -DCMAKE_INSTALL_PREFIX=$HOME/.local \
    -DCMAKE_PREFIX_PATH="$(pwd)/../deps/build/destdir/usr/local;/usr/local/lib64/cmake/Imath;/usr/local;/usr" \
    -DCMAKE_BUILD_TYPE=Debug

cmake --build . -- -j$(nproc)
```

Xlaits, 2025-12-09:
Thanks, this guide got me a successful build. They REALLY need to update their build docs and toolchains.

Xlaits, 2025-12-11:
> It's getting dated, so feel free to update.

Right now, it is working as of yesterday, with *no* issues, on Arch Linux. No need to update your guide at this time.

Probably could learn to and write up an AUR package using these instructions, though...

Makhaira, 2025-12-26:
I've managed to solve a a number of dependency hurdles trying to compile on CachyOS but I'm now blocked as it seems that the checks for IlmBase are fundamentally flawed since the changes to OpenVDB and OpenEXR mean that it's now redundant. Sadly my skills don't go as far as to patch this to get it working.

In case it helps others, using the guidance in this issue, coupled with cmake --graphviz to track down missing dependencies got me most of the way. This link (https://github.com/prusa3d/PrusaSlicer/issues/14899) helped me solve a Boost problem. Now I'm stuck at this Ilmbase point.

fpgirard, 2025-12-27:
instructions worked like a charm on Fedora 43 running on a Geekom A6 with AMD Ryzen.  I did have to dnf install automake and libtool but the rest was textbook.  Thanks for posting  these instructions.

fpgirard, 2025-12-31:
> > instructions worked like a charm on Fedora 43 running on a Geekom A6 with AMD Ryzen. I did have to dnf install automake and libtool but the rest was textbook. Thanks for posting these instructions.
> 
> Did you have to do anything about ilmbase?

Not that I recall - it was truly straightforward once i resolved those 2 dependencies.

themanyone, 2026-01-02:
If you're on Fedora, you can auto-install build deps. If the rpm builds, the repo should as well.

```
dnf download --source PrusaSlicer
sudo dnf builddep prusa-slicer-2.9.3-1.fc43.src.rpm
```

closed 2026-09-29 by prusaslicer-bot[bot]