prusa3d/PrusaSlicer #14868: Tree support will create segments on second layer, but will on the third layer (depending on the object some times heigher)

JonasApessos, 2025-09-18:
### Description of the bug

The tree support on some objects (especially when it has to create support under the object) will not create on the second layer it's first perimeter on trees that are inside an other tree. This will cause the printer to try to print in the air on the third layer where it did not create on the second layer.
This happens on the second layer after creating the base layer for adhesion.
It sometimes avoids the problem if it has to print from second layer to third layer by filling enough (using 0.4mm nose) but if i use 0.2mm it will cause severe problems from the start. On  other models it will start printing on the fourth layer...that cause huge issues if it needs support there...

### Project file & How to reproduce

I can't upload as it says there was an error...

1.Import object that may need support under the object's mass.
2.enable generate support
3.go to settings and change support type to tree.
4.slice object and look under big body's of masses.
5. try printing and see result for yourself.

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.9.3+flathub.org

### Operating system

Ubuntu 24.04

### Printer model

ender 3 pro

JonasApessos, 2025-10-01:
[image]

JonasApessos, 2025-10-02:
I've managed to reduce the size, please check the file. It should be half of a belt from the full model.

u89djt, 2025-10-02:
Yes, the new zip file you've added to the main post exhibits the problem you described. This screenshot shows the first extrusion of an organic support element that has nothing in the layer immediately beneath it.
I don't have anything to add. It may be that someone knows whether it should be linked to an existing issue.
[image]

closed 2026-09-29 by prusaslicer-bot[bot]