prusa3d/PrusaSlicer #14871: Support for post-processing scripts via INI files for custom printer profiles

Ferenc1234, 2025-09-18:
**Is your feature request related to a problem? Please describe.**
I am trying to add a custom printer in PrusaSlicer using an INI profile, but my printer requires a post-processing script to convert the generated G-code into a proprietary hexadecimal format. Currently, I could not find an option to link or embed a post-processing script directly in the INI file, which would make printer integration and workflow much easier.

**Describe the solution you'd like**
I would like PrusaSlicer to support specifying post-processing scripts directly within INI printer profiles. Ideally, this would allow me to select or write a script as part of the printer configuration, so that the G-code is automatically converted after slicing—without needing to manually add or configure the script each time.

**Describe how it would work**
- An option or field in the INI profile format to specify a post-processing script file or inline script.
- When a print is sliced for this printer, the script is run automatically on the output G-code, producing the required special format.
- This would streamline support for printers that need custom conversion, especially those using manufacturer-specific or proprietary file formats.

**Describe alternatives you've considered**
- Manually adding the post-processing script for each printer or print job, which is error-prone and inconvenient.
- Referring to other community profiles (such as the FlashForge CreatorPro2 PS Profile repo), but these still require manual script addition.

**Additional context**
Example what it could be used for: https://github.com/Jacotheron/FlashForge-CreatorPro2-PS-Profile

Thank you for considering this feature!

Ferenc1234, 2025-09-19:
@neophyl yes im fully aware of that. I stated that in the issue. It is impossible to add this new printer (using pull request), because it uses ini file

neophyl, 2025-09-19:
Sorry, correct me if I'm wrong but don't you also need to install a print profile to match your custom printer when its installed already ?  If so then that should allow you to embed the post processing script link into the print profile already. 
I guess what I'm questioning is why it needs to be moved to be part of the Printers profile and cant be achieved with its current location ?

closed 2026-09-29 by prusaslicer-bot[bot]