prusa3d/PrusaSlicer #14888: PLEASE Add print order:  inner perimeters → infill → outer perimeter

ZehnEuro, 2025-09-25:
When using MMU / multi-material prints, I often run into visible color bleed. For example, when switching from a very dark color to a light one, the dark pigment still contaminates the nozzle slightly. With the current print order (inner → outer → infill), this residue ends up in the outer perimeter, causing unwanted discoloration and surface artifacts. I already increased the recommended wipe volume by 500%, but the problem persists – and this also results in a lot of wasted filament. Using the existing option “Infill before perimeters” is also not a solution, because then the infill is printed before all perimeters, which leads to visible artifacts on the outer surface.

**Describe the solution you'd like**
I would like a new option to change the print order to:
_Inner perimeters -> Infill -> Outer perimeter (last)_
This way, any leftover material from the tool change would be purged into the infill first, and the final outer perimeter remains much cleaner.

**Describe how it would work**
It could be implemented as an additional checkbox in the “Print Settings → Layers and perimeters” section, e.g. “Print infill between inner and outer perimeters”. If enabled, the slicer would treat infill as part of the inner section, keeping the outer perimeter strictly last.

**Describe alternatives you've considered**
Increasing wipe volume: works partially but creates massive filament waste.
“Infill before perimeters”: solves color bleeding, but creates outer-surface artifacts since the outer wall no longer comes last.
Neither option solves the issue without major drawbacks.

**Additional context**
This feature would be especially beneficial for MMU users who want to print models with strong color contrasts (e.g. dark on light) without suffering from discoloration on the outside walls. It would improve both surface quality and reduce material waste.

vladimirusenko095-byte, 2025-09-30:
> ### 

- [ ] @ #- > [****](url)

n-dony, 2026-09-23:
@ZehnEuro try my fork :) https://github.com/n-dony/PrusaSlicer

closed 2026-09-29 by prusaslicer-bot[bot]