prusa3d/PrusaSlicer #14891: slicing without being told to

ronan-mcdonough, 2025-09-28:
### Description of the bug

prusa slicer is slicing w/o me telling it to, it begins anytime I make an edit before I have finished setting up the objects.

### Project file & How to reproduce

1) make any change to the objects
2) it just starts slicing w/o user telling it to

[stupid project that broke prusaslicer 2.9.3.zip](https://github.com/user-attachments/files/22584512/stupid.project.that.broke.prusaslicer.2.9.3.zip)

[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.3

### Operating system

windows 11

### Printer model

prusa mk4s 0.4 nozzle

ronan-mcdonough, 2025-09-28:
tried 2.9.2 and had same issue

rtyr, 2025-09-28:
Configuration --> Preferences. Disable option "Background processing".

ronan-mcdonough, 2025-09-28:
thank you, that worked.

closed 2025-09-28 by rtyr